riscv/sail-riscv #2: Use libfdt in the C emulator

pmundkur, 2019-01-22:
Currently, the FDTs for OS boots with the C emulator need to be manually checked for consistency against the constants defined in the C platform.  It would be better for these values to be derived directly from the FDT, using libfdt from the device-tree-compiler package.  This would add a dependency to the C emulator, but libfdt is a simple dependency to satisfy on Linux systems.

allenjbaum, 2021-10-01:
This issue is 2 1/2 years old. IS it still valid?

Alasdair, 2021-10-01:
I think this still falls into the nice-to-have category, but nobody every got around to implementing it.

allenjbaum, 2021-10-02:
OK - but does this mean the issue should remain open, or be closed? I'm not sure of the etiquette here.
I think of "issues" meaning problems, not wishlists. IF we could tag it with "wishlist", that would be OK too, I suppose.
Note that will this may be "a simple dependency to satisfy on a Linux system",  not all systems are Linux.

martinberger, 2021-10-02:
> IF we could tag it with "wishlist",

Do we have a place where we put wishes and the like?

allenjbaum, 2021-10-04:
The arch-test repo uses the label dropdown on the right side of the issue
page.
It is populated in that repo with:
bug.                       Something isn't working
duplicate               This issue or pull request already exists
enhancement.        New feature or request
fixed-in-RISCOF.   (shorthand for: will be fixed in next major revision)
good first issue.     Good for newcomers
help wanted.          Extra attention is needed
invalid                    This doesn't seem right, or maybe not-a-bug
question                 Further information is requested
wontfix

I'm not asking for all of these, but bug, duplicate, enhancement,
not-a-bug,  is a good start.

martinberger, 2021-10-04:
>  I'm not asking for all of these, but  ...

Good idea. This is supported by Github anyway.,

pmundkur, 2025-06-13:
Closing since this is subsumed by #993.

closed 2025-06-13 by pmundkur