riscv/sail-riscv #4: Overflow when comparing addresses in riscv_platform.sail

PeterRugg, 2019-04-12:
(As discussed with @rmn30, found using TestRIG)

Some of the within... functions, e.g. within_phys_mem, have an overflow issue when accessing addresses very near the top of the address space.

Here is a check inside within_phys_mem:
`if (  ram_base <=_u ext_addr
& (ext_addr + sizeof('n)) <=_u (ram_base + ram_size))`

Here, if ext_addr is 0xffff_ffff (assuming 32 bit memory addresses), and sizeof ('n) is (e.g.) 1, this access will show as being within physical memory regardless of the values of ram_base and ram_size, since in all cases ram_base <=_u 0xffff_ffff, and 0xffff_ffff + 1 = 0 <=_u ram_base + ram_size due to overflow.

Fix should be straightforward (e.g. also check if ext_addr + sizeof('n) >= ext_addr), but I'm not sure in how many places this is an issue. It seems to also be a problem in within_clint.

pmundkur, 2019-04-12:
Thanks for finding this.  It was fixed in f62cc79dd53bedb3dea1ea9c0d0b965b4fc1790a in the cheri-merge branch, which we're planning to merge into master fairly soon.

pmundkur, 2019-05-10:
Fix was merged to master some time ago. closing

closed 2019-05-10 by pmundkur