riscv/sail-riscv #7: Build error

ccasin, 2019-05-25:
After installing sail, I ran `make` from top-level `sail-riscv` directory.  It failed fairly quickly with an SMT-related error:

```
mkdir -p generated_definitions/ocaml/RV64
sail  -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV64 -o riscv model/prelude.sail model/prelude_mapping.sail model/riscv_xlen64.sail model/prelude_mem_metadata.sail model/prelude_mem.sail model/riscv_types.sail model/riscv_reg_type.sail model/riscv_regs.sail model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_csr_map.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail model/riscv_next_control.sail model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv39.sail model/riscv_vmem_sv48.sail model/riscv_vmem_rv64.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail model/riscv_insts_next.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail model/riscv_step.sail model/riscv_analysis.sail
Error:
Error when calling smt: Reporting.Fatal_error(_)
Makefile:147: recipe for target 'generated_definitions/ocaml/RV64/riscv.ml' failed
make: *** [generated_definitions/ocaml/RV64/riscv.ml] Error 1
```

The error above is with ocaml 4.06.1.  I also tried 4.07.1, and got this similar but slightly different error:

```
mkdir -p generated_definitions/ocaml/RV64
sail  -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV64 -o riscv model/prelude.sail model/prelude_mapping.sail model/riscv_xlen64.sail model/prelude_mem_metadata.sail model/prelude_mem.sail model/riscv_types.sail model/riscv_reg_type.sail model/riscv_regs.sail model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_csr_map.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail model/riscv_next_control.sail model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv39.sail model/riscv_vmem_sv48.sail model/riscv_vmem_rv64.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail model/riscv_insts_next.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail model/riscv_step.sail model/riscv_analysis.sail
Error:
SMT solver returned unexpected status 110
Makefile:147: recipe for target 'generated_definitions/ocaml/RV64/riscv.ml' failed
make: *** [generated_definitions/ocaml/RV64/riscv.ml] Error 1
```

Googling the error didn't turn up anything - any help would be appreciated.  I'm building on Ubuntu 18.04 under WSL.

ccasin, 2019-05-25:
As an aside, `z3` is installed and seems to be working from the command line, though I'm not sure if that's the SMT solver `sail` is attempting to use.

Alasdair, 2019-05-25:
Seems strange. I've never used WSL, but I do use the default z3 on Ubuntu 18.04 myself and it works there. There is a SAIL_FLAGS setting in the makefile you could use to set the "-dsmt_verbose" option which prints all generated smt problems and attach that output along with the z3 version. You could try using a newer version of z3 and see if that changes anything. While Sail should use z3 by default there is a (very experimental) sail flag (-smt_solver) to use a different smt solver, so you could try using cvc4, which did work for sail-riscv last time I tried it.

The difference between the two errors looks like the ocaml compiler bug we believed fixed in the 4.06.1 point release where it miscompiles pattern matching on exceptions.

ccasin, 2019-05-25:
I got the additional output below with that flag.  I tried copying the listed constraints to a file and calling `z3` directly, and it seems to work (returns `unsat`).  I'll try from a native ubuntu machine tonight or tomorrow to see if it's a WSL issue.

```
mkdir -p generated_definitions/ocaml/RV64
sail -dsmt_verbose -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV64 -o riscv model/prelude.sail model/prelude_mapping.sail model/riscv_xlen64.sail model/prelude_mem_metadata.sail model/prelude_mem.sail model/riscv_types.sail model/riscv_reg_type.sail model/riscv_regs.sail model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_csr_map.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail model/riscv_next_control.sail model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv39.sail model/riscv_vmem_sv48.sail model/riscv_vmem_rv64.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail model/riscv_insts_next.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail model/riscv_step.sail model/riscv_analysis.sail
Prove (type_check.ml/2135)  |- true
Prove (type_check.ml/2135)  |- ((not(not('n == 'm)) | 'n != 'm) & (not('n != 'm) | not('n == 'm)))
SMTLIB2 constraints are: 
(push)
(declare-const v0 Int)
(declare-const v1 Int)
(define-fun constraint () Bool (not (and (or (not (not (= v1 v0))) (not (= v1 v0))) (or (not (not (= v1 v0))) (not (= v1 v0))))))
(assert constraint)
(check-sat)
(pop)

Error:
SMT solver returned unexpected status 110
Makefile:148: recipe for target 'generated_definitions/ocaml/RV64/riscv.ml' failed
make: *** [generated_definitions/ocaml/RV64/riscv.ml] Error 1
```

Alasdair, 2019-05-25:
I had a look and 110 in seems like an internal z3 error of some kind

```
#define ERR_INTERNAL_FATAL      110
```

ccasin, 2019-05-26:
I tried this on a non-virtualized ubuntu machine and did not receive the same error, so it's probably either a problem with WSL or an issue with the specific version of z3 or other packages in my WSL ubuntu instance.  I'd like to investigate more, since WSL-specific errors are quite rare these days in my experience, but I'm closing the issue for now.  Thanks for your help.

closed 2019-05-26 by ccasin