riscv/sail-riscv #16: sstatus/sie/sip don't really exist

scottj97, 2019-07-24:
# Problem

There is a difference between Sail and Spike w.r.t. the CSRs sstatus, sie, and sip.

Currently the RISC-V Sail model reports registers sstatus, sie, and sip as CSRs when they get written by a CSRW (et al) instruction, but these registers are actually implemented using the mstatus, mie, and mip registers.

Spike reports the actual underlying registers that are updated: mstatus, mie, and mip.

# Explanation

The privileged ISA section 4.1.1 (sstatus) has commentary about this: "The sstatus register is a subset of the mstatus register. In a straightforward implementation, reading or writing any field in sstatus is equivalent to reading or writing the homonymous field in mstatus."

# Current solution

The [trace compare program](https://github.com/rems-project/sail-riscv/blob/7e6ffe27372f6d740e3f96ec94a3dc05047b5500/ocaml_emulator/tracecmp.ml#L298) has special-case code to deal with these registers, since Spike handles them differently.

# Proposed solution

I think it makes more sense for the Sail model to report the actual register that was updated: mstatus, mie, or mip, instead of the supervisor-level shadow registers. This way, the comparison will not have to be special-cased.

To achieve this I added the following code right above the print_reg in riscv_insts_zicsr.sail:

```
      /* These three supervisor CSRs are implemented as a subset of the machine CSRs,
         so when they are printed I want to see the machine CSR names. */
      let printcsr : csreg =
      match csr {
        0x100 => 0x300,  /* sstatus -> mstatus */
        0x104 => 0x304,  /* sie     -> mie     */
        0x144 => 0x344,  /* sip     -> mip     */
        _     => csr
      };
```

# Questions

Is this a worthwhile goal?

Is this the best way to achieve it?

allenjbaum, 2021-02-26:
I personally have no problem if the shadow registers are reported; I don't think any tool will care.
There is only one physical register with 3 views, and the report should correctly identify view that was accessed - that is what the programmer will see at that address.

scottj97, 2021-02-26:
Well, I have an internal tool that cares, which is how I found this problem in the first place.

I consider the trace to be a recording of state changes. Since there is no state named sstatus, only mstatus, it doesn't make sense to report a change to a register named sstatus.

allenjbaum, 2021-10-01:
revisiting this:

I'm unclear about what the internal tool is doing with this information, or if it should even be doing something with that (extra) information.
Is this to just remain consistent with Spike (vs. doing it the other way around)? Or is the some other reason?

allenjbaum, 2021-10-01:
Just talked to the riscof team who prefer your solution - but would like to add versioning first so that incompatible changes can be tracked (and this will make an incompatibility with their tools, thought its easy to fix and they prefer it).

jrtc27, 2021-10-01:
The current trace output it not a stable interface, changing it is fine

allenjbaum, 2021-10-01:
I'm not sure what the implications of it becoming "stable" means.
Bit, as I mentioned elsewhere, if there is SW that depends on it - should
it become stable? (or at least versioned)

scottj97, 2021-10-01:
My internal tool is doing an instruction-by-instruction state compare vs another implementation (such as RTL Verilog or Spike).

For that purpose, I want to compare state (like `mstatus`), not software views (like `sstatus`).

Another use case: looking through a trace, trying to understand how/where _e.g._ `mstatus.SUM` got set to 1. That bit can be accessed through two different CSR names, but there is only one bit of state. It's confusing to have to search for two different register names in the trace to find all places where that one bit changes.

There are many more such cases in the hypervisor extension.

allenjbaum, 2021-10-02:
That's pretty much what I thought was going on. The riscof team has the same issue.
I'm fine with this.
Want to submit a pull request for it?

jrtc27, 2021-10-02:
I don't see why riscof should care. The log format is not something that should be baked into it, for that you want a very clearly-defined machine-readable trace, not the human-readable thing that currently is emitted.

jrtc27, 2021-10-02:
(isn't there a trace document in the works for precisely that kind of thing?)

jrtc27, 2021-10-02:
There are also a couple of problems with the patch as it stands:
1. The remapping is kept separate from the actual implementation, so it's easy to add a new CSR and forget to remap it. This is going to be particularly relevant for the H extension where there are a lot of S->VS CSR remappings going on.
2. There needs to be an ext hook so any CSRs defined by out-of-tree extensions are able to remap theirs similarly.

It might end up being too unwieldy, but you can kill two birds with one stone by changing res (and ext_write_CSR's return type) from `option(xlenbits)` to `option(xlenbits, csreg)` or `option(xlenbits, option(csreg))` (the latter is maybe conceptually more elegant, but `csr` is shorter to write than `None()`).

scottj97, 2021-10-02:
I agree with @jrtc27's criticisms. I don't know the Sail language that well which is why I offered this as a proposal instead of PR.

allenjbaum, 2021-10-02:
Riscof is using what is available, and it works well enough.

The RVI Trace format is designed for maximum compressibility, because it
will be driven off-chip or into some RAM buffer.
This is reasonably complex to encode and decode - and not applicable to a
SW model that doesn't have those BW restrictions.

I won't claim to understand where the code that generates that trace
resides (model specific or built into the language),
but if it's built into our RISC-V C-model: we control it, and we should
choose the format that makes our life easiest.
Using what is there is easiest, and if a tweak makes something even easier,
we should tweak it.

That says nothing about how this specific change should be made, I hasten
to add; I have no opinions on the patch.