riscv/sail-riscv #17: Cannot Build

eroom1966, 2019-09-05:
I have installed sail (succesfully I think)
when trying to run 'make' for sail-riscv I getthe following errors
any help appreciated

> $ ARCH=RV32 make
> mkdir -p generated_definitions/ocaml/RV32
> sail  -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV32 -o riscv model/prelude.sail model/prelude_mapping.sail model/riscv_xlen32.sail model/prelude_mem_metadata.sail model/prelude_mem.sail model/riscv_types.sail model/riscv_reg_type.sail model/riscv_regs.sail model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_pmp_regs.sail model/riscv_pmp_control.sail model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_csr_map.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail model/riscv_next_control.sail model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv32.sail model/riscv_vmem_rv32.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail model/riscv_insts_next.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail model/riscv_step.sail
> Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:76:7-12
> 76 |  then match n {
>    |       ^---^
>    | 
> Most general matched pattern is 0
> 
> Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:82:7-12
> 82 |  else match n { //  sizeof(xlen) == 64
>    |       ^---^
>    | 
> Most general matched pattern is 0
> 
> Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:95:7-12
> 95 |  then match n {
>    |       ^---^
>    | 
> Most general matched pattern is 0
> 
> Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:118:7-12
> 118 |  then match n {
>     |       ^---^
>     | 
> Most general matched pattern is 0
> 
> cp: cannot stat '/home/moore/.opam/4.06.1/share/sail/src/value.ml': No such file or directory
> cp: cannot stat '/home/moore/.opam/4.06.1/share/sail/src/toFromInterp_lib.ml': No such file or directory
> mkdir -p ocaml_emulator/_sbuild
> cp ocaml_emulator/_tags ocaml_emulator/platform.ml ocaml_emulator/platform_impl.ml ocaml_emulator/riscv_ocaml_sim.ml generated_definitions/ocaml/RV32/*.ml ocaml_emulator/_sbuild
> cd ocaml_emulator/_sbuild && ocamlbuild -use-ocamlfind riscv_ocaml_sim.native
> + ocamlfind ocamldep -package lem -package linksem -package zarith -modules riscv_ocaml_sim.ml > riscv_ocaml_sim.ml.depends
> ocamlfind: Package `lem' not found
> Command exited with code 2.
> Compilation unsuccessful after building 1 target (0 cached) in 00:00:00.
> Makefile:162: recipe for target 'ocaml_emulator/_sbuild/riscv_ocaml_sim.native' failed
> make: *** [ocaml_emulator/_sbuild/riscv_ocaml_sim.native] Error 10

rmn30, 2019-09-05:
Hello! 

How did you install sail and lem? It looks like you used opam. If so then ocamlfind should be able to find lem so this is strange. The warnings are nothing to worry about and even the cp errors are normal and harmless. Please could you give the result of running `opam list`? Does `/home/moore/.opam/4.06.1/lib` contain a `lem` directory?

Incidentally, do you want an emulator? If so then you might be better off just building the C emulator with `ARCH=RV32 make c_emulator/riscv_sim_RV32` . The default target will build ocaml and C emulators as well as theorem prover files.

eroom1966, 2019-09-11:
> Does /home/moore/.opam/4.06.1/lib contain a lem directory?

Yes it does, although the contents are only 2 files
META
opam.config  ( this file is zero bytes)
is this correct ?

> Incidentally, do you want an emulator?

What I want is the version of the simulator which is intended to be adopted as the golden reference by the foundation ?
are the C and ocaml references equivalent ?
If so, then yes I will use the C emulator, can you refer me to the full set of build instructions for the C emulator ?
Thx
Lee

eroom1966, 2019-09-11:
Update
actually running the command
> ARCH=RV32 make c_emulator/riscv_sim_RV32

Seems to have created this executable, I will try with this to see how far I get
Thx
Lee

rmn30, 2019-09-11:
> > Does /home/moore/.opam/4.06.1/lib contain a lem directory?

> Yes it does, although the contents are only 2 files
> META

This is the important bit. On my system that file contains the following:
```
version = ""
description = "Lem library for extracted OCaml"
requires = "lem_zarith"
requires(nozarith) = "lem_num"
```
This means that ocamlfind should look for `lem_zarith` package unless `nozarith` is specified. Zarith is needed to support arbitrary precision integers and uses the GMP library. If you also have a lem_zarith package and libgmp is installed I don't know why ocamlfind isn't finding it. Perhaps `ocamlfind query -r lem` will provide a clue?

> opam.config ( this file is zero bytes)

I don't have this but I don't think it is harmful. It might be an `opam 1.2` vs `opam 2` difference (I'm using `opam 2`).

> Seems to have created this executable, I will try with this to see how far I get

Glad to be of help.

eroom1966, 2019-09-11:
@rmn30 
Thanks for your help, I have managed to build the C Emulator AND successfully run the riscv-compliance suite and matched the signatures - great!

I wanted to get some trace output, but cannot quite work out how to do this.
I see there is a command line option --trace
so what I was hoping was to possibly generate the following 
- disassembly
- register changes
- memory load/stores

should trace provide any of these capabilities ?

thx
Lee

eroom1966, 2019-09-11:
Where does it output ?
I ran strace and could see writes to FD=1 and FD=2 (stdout, stderr)
but this only consisted of a series of '\n'
so all I get is a series of newlines
thx
Lee

rmn30, 2019-09-11:
It should be output on stdout when the emulator is run. This sounds a bit familiar: there used to be a bug in sail optimisation that meant the print statements were optimised away. Please try either removing  `-Oconstant_fold` in the [Makefile here](https://github.com/rems-project/sail-riscv/blob/d07ded0537e9367bb6e8c53f75a02403b6cdc0e4/Makefile#L199) or changing `false` to `true` in the sail-riscv [prelude here](https://github.com/rems-project/sail-riscv/blob/d07ded0537e9367bb6e8c53f75a02403b6cdc0e4/model/prelude.sail#L111) and recompile. The sail bug was fixed a while ago but we might need to release a new opam package.

rmn30, 2019-09-11:
Yes, unfortunately [this sail commit](https://github.com/rems-project/sail/commit/01177094d660286ebe8c83c68d1e8890e29d154c) fixed the bug just after the last release. I will release a new opam package soon. In the meantime turning off `-Oconstant_fold` as above should work or you could try using opam pin to [get the latest development version of sail](https://github.com/rems-project/sail/wiki/OPAMInstall#installing-development-versions-of-sail). Sorry for the confusion.

eroom1966, 2019-09-11:
> In the meantime turning off -Oconstant_fold as above should work 

This worked
Thx
Lee

closed 2020-06-18 by eroom1966