riscv/sail-riscv #29: FENCE instructions should ignore rs1/rd and address fm=2..15 to same as fm=0

ss2783, 2019-12-18:
The sail-riscv model seems like it expects the rs1/rd fields to be zeros and only defines instruction variants for the FENCE instruction with fm = 0 and fm = 1 (TSO). 

From the ISA manual volume I, section 2.7 on Memory Ordering Instructions states: 

"The unused fields in the FENCE instructions—rs1 and rd—are reserved for finer-grain fences in future extensions. For forward compatibility, base implementations shall ignore these fields, and standard software shall zero these fields. Likewise, many fm and predecessor/successor set settings
in Table 2.2 are also reserved for future use. Base implementations shall treat all such reserved configurations as normal fences with fm=0000, and standard software shall use only non-reserved
configurations"

These sound more like software restrictions on how the reserved field should be used but the sail model should ignore rs1/rd fields. Likewise fm=2..15 should be the same as fm=0

rmn30, 2019-12-19:
@pmundkur See above for proposed fix for this on branch fence_issue29.

scottj97, 2020-08-28:
I just encountered this issue myself, so giving it a bump.

@rmn30's patch on branch fence_issue29 seems incomplete, because attempts to execute a FENCE with those fields nonzero results in a failure:
```
Pattern match failure in assembly_forwards
```
(As already discussed with @pmundkur in 1a6d57a.)

There also seems to be a similar problem with FENCE.I:

> The unused fields in the FENCE.I instruction, imm[11:0], rs1, and rd, are reserved for finer-grain
fences in future extensions. For forward compatibility, base implementations shall ignore these
fields, and standard software shall zero these fields.

pmundkur, 2020-09-04:
I pushed a fix for handling hints that should address this.

scottj97, 2020-09-07:
Fix works for me.

pmundkur, 2020-09-09:
Great, thanks @scottj97.  @ss2783, can i close this?

closed 2020-10-16 by pmundkur