riscv/sail-riscv #46: Build fails on all Archs

ShvadronUzi, 2020-03-30:
I am getting the following error on the current sail-riscv version 
(I added SAIL_FLAGS += -dsmt_verbose to the Makefile):
========================
$ make
mkdir -p generated_definitions/ocaml/RV64
sail -dno_cast -dsmt_verbose -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV64 -o riscv model/prelude.sail model/prelude_mapping.sail model/riscv_xlen64.sail model/riscv_flen_D.sail model/prelude_mem_metadata.sail model/prelude_mem.sail model/riscv_types_common.sail model/riscv_types_ext.sail model/riscv_types.sail model/riscv_vmem_types.sail model/riscv_reg_type.sail model/riscv_freg_type.sail model/riscv_regs.sail model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_pmp_regs.sail model/riscv_pmp_control.sail model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_misa_ext.sail model/riscv_csr_map.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail model/riscv_next_control.sail model/riscv_fdext_regs.sail model/riscv_fdext_control.sail model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_pte.sail model/riscv_ptw.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv39.sail model/riscv_vmem_sv48.sail model/riscv_vmem_rv64.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail model/riscv_insts_next.sail model/riscv_softfloat_interface.sail model/riscv_insts_fext.sail model/riscv_insts_dext.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail model/riscv_step.sail model/riscv_analysis.sail
Prove (type_check.ml/2207)  |- ((not(not('n == 'm)) | 'n != 'm) & (not('n != 'm) | not('n == 'm)))
SMTLIB2 constraints are: 
(push)
(declare-const v0 Int)
(declare-const v1 Int)
(define-fun constraint () Bool (not (and (or (not (not (= v1 v0))) (not (= v1 v0))) (or (not (not (= v1 v0))) (not (= v1 v0))))))
(assert constraint)
(check-sat)
(pop)

Error:
Error when calling smt: Reporting.Fatal_error(_)
make: *** [generated_definitions/ocaml/RV64/riscv.ml] Error 1

==================
I have seen a reference to such an eror before but it seems that this is a source problem which has not been resolved

bauereiss, 2020-03-30:
What versions of Sail and z3 do you have installed?

ShvadronUzi, 2020-03-30:
On WSL Ubuntu:
$ sail -v
Sail 0.12 (sail2 @ opam)
$ z3 -version
Z3 version 4.4.1

On RH6 same Sail  but Z3 is not found. Nevertheless, the error is the same

Alasdair, 2020-03-30:
If you try creating a .smt2 file with
```
(push)
(declare-const v0 Int)
(declare-const v1 Int)
(define-fun constraint () Bool (not (and (or (not (not (= v1 v0))) (not (= v1 v0))) (or (not (not (= v1 v0))) (not (= v1 v0))))))
(assert constraint)
(check-sat)
(pop)
```
And invoking z3 directly on that file does it work?

We've had reports in the past that z3 doesn't work on WSL (then it failed with an undocumented error code >100), and the issue went away when they switched to an actual Linux machine.

> On RH6 same Sail but Z3 is not found. Nevertheless, the error is the same

I think the error may be the same but I guess the reason is likely different: z3 needs to be installed and available in $PATH. I should improve the error message for this particular case.

ShvadronUzi, 2020-03-31:
On WSL Ubuntu:
$ z3 tst.smt2
unsat

On RH6, I am waiting for IT to install z3... will update here

Alasdair, 2020-03-31:
Something strange is happening. The code that creates that exception looks like:
```
       try
         let smt_out, smt_in, smt_err = Unix.open_process_full (!opt_solver.command ^ " " ^ input_file) (Unix.environment ()) in
         let smt_output =
           try List.combine problems (input_lines smt_out (List.length problems)) with
           | End_of_file -> List.combine problems ["unknown"]
         in
         let status = Unix.close_process_full (smt_out, smt_in, smt_err) in
         status, smt_output
       with
       | exn ->
          raise (Reporting.err_general l ("Error when calling smt: " ^ Printexc.to_string exn))
```
It only calls OCaml standard library functions (List.*, Unix.*, and input_line via input_lines), yet in the Error message:
```
Error when calling smt: Reporting.Fatal_error(_)
```
The exception being caught is defined within Sail, so it shouldn't be throwable from that code. OCaml 4.06 had a bad miscompilation bug where it would compile exceptions handlers incorrectly that caused a similar issue. I really hope that hasn't regressed - what version of OCaml do you have?

ShvadronUzi, 2020-03-31:
$ ocaml -version
The OCaml toplevel, version 4.06.1

Alasdair, 2020-03-31:
Anything greater than 4.06.1 should work.

I've tried to reproduce your environment as closely as I can (same z3, Sail, and ocaml), but I can't reproduce this. Even if I inject failures into the parts of the function that can raise Reporting.Fatal_error I get:
```
[alasdair sail-riscv]$ export TMPDIR=/root
[alasdair sail-riscv]$ make
...
Error:
Could not open temp file when calling SMT: /root/constraint_41721e.smt2: Permission denied
```
and when symlinking z3 to /usr/bin/false
```
[alasdair sail-riscv]$ make      
Error:
SMT solver returned unexpected status 1
```
which are the error messages I would expect

ShvadronUzi, 2020-03-31:
The current INSTALL.md states:
opam switch 4.06.1
You might consider changing it to:
opam switch 4.07.0 (or higher)
Should I just do the switch or continue with the rest of the (re)installation?

Alasdair, 2020-03-31:
4.06.1 is the minimum version we test with (our Jenkins setup uses it right now), but if you can use a higher version like 4.07 or 4.08 I'd recommend it.

ShvadronUzi, 2020-03-31:
I switched to 4.07.0 and reinstalled sail (just to make sure... as I was not told otherwise).
I am now get slightly more information on the same issue:
Prove (type_check.ml/2207)  |- ((not(not('n == 'm)) | 'n != 'm) & (not('n != 'm) | not('n == 'm)))
SMTLIB2 constraints are:
(push)
(declare-const v0 Int)
(declare-const v1 Int)
(define-fun constraint () Bool (not (and (or (not (not (= v1 v0))) (not (= v1 v0))) (or (not (not (= v1 v0))) (not (= v1 v0))))))
(assert constraint)
(check-sat)
(pop)

Error:
**SMT solver returned unexpected status 110**
Makefile:183: recipe for target 'generated_definitions/ocaml/RV64/riscv.ml' failed
make: *** [generated_definitions/ocaml/RV64/riscv.ml] Error 1

Alasdair, 2020-03-31:
Yes, that's the error code I've seen before from another person using WSL

Alasdair, 2020-03-31:
I'd be curious to know what causes it, because it'd be nice if it worked, but I probably can't get access to a windows machine right now.

ronorton, 2020-03-31:
I'm able to reproduce this under WSL. Z3 on above smt file exits with code 0 so maybe it is an ocaml or WSL bug?

ronorton, 2020-03-31:
strace seems to show z3 exiting with code 110 for some reason:
```
openat(AT_FDCWD, "/tmp/constraint_cf7eba.smt2", O_WRONLY|O_CREAT|O_EXCL|O_CLOEXEC, 0600) = 3
lseek(3, 0, SEEK_CUR)                   = 0
write(3, "(push)\n(declare-const v0 Int)\n(d"..., 221) = 221
close(3)                                = 0
pipe2([3, 4], O_CLOEXEC)                = 0
pipe2([5, 6], O_CLOEXEC)                = 0
pipe2([7, 8], O_CLOEXEC)                = 0
lseek(3, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
lseek(6, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
lseek(7, 0, SEEK_CUR)                   = -1 ESPIPE (Illegal seek)
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7f46914d0e50) = 1146
close(5)                                = 0
close(4)                                = 0
close(8)                                = 0
read(3, "", 65536)                      = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=1146, si_uid=1000, si_status=110, si_utime=0, si_stime=0} ---
close(3)                                = 0
close(6)                                = 0
close(7)                                = 0
wait4(1146, [{WIFEXITED(s) && WEXITSTATUS(s) == 110}], 0, NULL) = 1146
write(2, "Error:\n", 7Error:
)                 = 7
write(2, "SMT solver returned unexpected s"..., 42SMT solver returned unexpected status 110
) = 42
exit_group(1)                           = ?
+++ exited with 1 +++
```

bacam, 2020-04-01:
That appears to be its other-internal-error return code.  I'll try to make Sail capture stderr so that we can see what the error message was.

Alasdair, 2020-04-01:
Yes, we should capture stderr in these cases.

@ronorton I made a little test file that mimics what Sail is doing. Maybe you could try it on WSL? It should compile and run with just `ocamlc unix.cma test.ml && ./a.out`.
```
exception No_temp_file
exception Unexpected_status of int
exception Killed_by_signal of int

let smt_file =
  "(push)\n\
   (declare-const v0 Int)\n\
   (declare-const v1 Int)\n\
   (define-fun constraint () Bool (not (and (or (not (not (= v1 v0))) (not (= v1 v0))) (or (not (not (= v1 v0))) (not (= v1 v0))))))\n\
   (assert constraint)\n\
   (check-sat)\n\
   (pop)\n"

let rec input_lines chan = function
  | 0 -> []
  | n ->
     let l = input_line chan in
     let ls = input_lines chan (n - 1) in
     l :: ls

let read_chan chan =
  let lines = ref [] in
  try
    while true; do
      lines := input_line chan :: !lines
    done; !lines
  with End_of_file ->
    List.rev !lines
 
let _ =
  let (input_file, tmp_chan) =
    try Filename.open_temp_file "constraint_" ".smt2" with
    | Sys_error msg -> raise No_temp_file
  in
  output_string tmp_chan smt_file;
  close_out tmp_chan;
  let status, smt_output, smt_errors =
    try
      let smt_out, smt_in, smt_err = Unix.open_process_full ("z3 -t:1000 -T:10 " ^ input_file) (Unix.environment ()) in
      let smt_output =
        try input_lines smt_out 1 with
        | End_of_file -> ["unknown"]
      in
      let smt_errors = read_chan smt_err in
      let status = Unix.close_process_full (smt_out, smt_in, smt_err) in
      status, smt_output, smt_errors
    with
    | exn -> raise exn
  in
  let _ = match status with
    | Unix.WEXITED 0 -> ()
    | Unix.WEXITED n ->
       List.iter prerr_endline smt_errors;
       raise (Unexpected_status n)
    | Unix.WSIGNALED n | Unix.WSTOPPED n ->
       List.iter prerr_endline smt_errors;
       raise (Killed_by_signal n)
  in
  Sys.remove input_file;
  List.iter print_endline smt_output
```

davidchisnall, 2020-04-01:
Newer versions of Z3 are tested with WSL and expected to work, but 4.4.1 is very old (4.8.7 is the latest, 4.4.1 is from 2015).  Please can you try with a newer version?

bacam, 2020-04-01:
Just pushed a change to report stderr.

ronorton, 2020-04-01:
I downloaded the 4.8.7 release of z3 for ubuntu-16.04 from https://github.com/Z3Prover/z3/releases and added bin/z3 to path before running sail and the error went away :-)

Alasdair, 2020-04-01:
Ok that's good to hear! I'll add a note to our install instructions later about that.

ShvadronUzi, 2020-04-01:
Z3 is installed via opam... So the package should be changed there, right? Should I wait for update in github?

Alasdair, 2020-04-01:
It's not an opam package, we typically suggest installing it via apt-get or any other package manager, but it seems like it might be that on WSL the package that comes in the ubuntu repositories doesn't work correctly as it's quite old. The link that Robert posted to the z3 releases page has binary downloads for newer versions of z3.

ShvadronUzi, 2020-04-01:
OK, I installed the latest Z3 (4.8.8) and indeed the issue above has been resolved. And -
The next error is:
Finished, 25 targets (25 cached) in 00:00:35.
ln: failed to create symbolic link '/mnt/g/RISC_V_tools/sail/sail-riscv/ocaml_emulator/_sbuild/riscv_ocaml_sim.native': Operation not permitted
rm -f ocaml_emulator/riscv_ocaml_sim_RV64 && cp -L ocaml_emulator/_sbuild/riscv_ocaml_sim.native ocaml_emulator/riscv_ocaml_sim_RV64 && rm -f ocaml_emulator/_sbuild/riscv_ocaml_sim.native
cp: cannot stat 'ocaml_emulator/_sbuild/riscv_ocaml_sim.native': No such file or directory
Makefile:198: recipe for target 'ocaml_emulator/riscv_ocaml_sim_RV64' failed
make: *** [ocaml_emulator/riscv_ocaml_sim_RV64] Error 1
===============
riscv_ocaml_sim.native does not exist at sail-riscv/ocaml_emulator/_sbuild

Alasdair, 2020-04-01:
Looks like some kind of permissions issue in your /mnt/g/ directory?

I know that windows requires a special UAC privilege to create symlinks, so if /mnt/g/ is how WSL mounts a windows filesystem (`G:\`?) then I could see that causing issues perhaps? That is mostly speculation on my part though.

ShvadronUzi, 2020-04-02:
OK, I now have sail-riscv working and I am able to run several tests on WSL :-)
To summarize what solved the build issue:
opam switch 4.07.0
Z3 4..8.8
The last problem was indeed a local file system issue

closed 2020-04-02 by ShvadronUzi