riscv/sail-riscv #50: N-extension CSRs active despite MISA

scottj97, 2020-05-21:
When `misa[13]==0` (Sail's default) the N-extension (user interrupts) is not present.

I observe that some of the N-extension behavior is properly dependent on `misa` (via `haveNExt()`) but not all.

# Expected behavior

Access to the N-extension CSRs (e.g. `ustatus`) should cause an illegal instruction exception.

Bits `mstatus[4]` (aka UPIE) and `mstatus[0]` (aka UIE) should be hardwired to 0.

# Actual behavior

No illegal instruction exception occurs.

UIE and UPIE are writable and retain the value last written.

scottj97, 2020-05-21:
Note that in the older (ratified) privileged spec, the description of mstatus expressly states that UPIE and UIE are hardwired to 0 in the absence of the N extension.

In the latest privileged spec, those two bits are marked as "WPRI".

The Sail model can be compatible with both versions of the spec by hardwiring them to 0.

pmundkur, 2020-05-22:
Great catch. I've pushed a fix to branch fix_next_csrs.  Do you have a test case for this?

scottj97, 2020-05-22:
I don't have a good test case, but I confirm this fixes the issue I had observed.

commit 710d499c (2020-05-23): Prevent access to N-mode registers and mstatus/mip/mie bits when N-mode is disabled.
Fixes #50.
closed 2020-05-23 by pmundkur