riscv/sail-riscv #56: PTW_Ext_Error is not used as a constructor

nwf, 2020-05-29:
The translation machinery in `riscv_vmem_sv39.sail`, for example, maps all `checkPTEPermissions` failures to `PTE_No_Permission`...
https://github.com/rems-project/sail-riscv/blob/fa707da94d0329e7d1b7520d876ad85953f2fdd5/model/riscv_vmem_sv39.sail#L43-L46
https://github.com/rems-project/sail-riscv/blob/fa707da94d0329e7d1b7520d876ad85953f2fdd5/model/riscv_vmem_sv39.sail#L120-L121
... which seems somewhat suspect.  Perhaps the correct thing to do would be to change
https://github.com/rems-project/sail-riscv/blob/f9f066e89a5c20f2769fbfb2b6a585af9b3f0561/model/riscv_pte.sail#L31-L34
to
```
union PTE_Check = {
  PTE_Check_Success : ext_ptw,
  PTE_Check_Failure : PTW_Error
}
```
This probably also obviates the `ext_ptw` field within the `TR_Failure` constructor?

pmundkur, 2020-05-29:
Making `PTE_Check` depend on `PTW_Error` creates a dependency problem between `riscv_pte.sail` and `riscv_ptw.sail`.  If it is not possible to isolate them with only a one-way dependency, we might need to fold both into a single file.  It might be simpler that way too, since someone extending `pte` stuff will almost surely also want to extend `ptw`, so having them both in a single file makes sense.

nwf, 2020-05-29:
Would it suffice to parameterize `PTE_Check` on its failure payload, so that the dependence on `PTW_Error` can be specified from outside?

pmundkur, 2020-05-29:
It would, but that outside would still have to be in some file that is extension specific (right?), which would add another file for the no-extension case.  It might be simpler to have the extension control {`riscv_pte`+`riscv_ptw`} and have `riscv_vmem_*` pick them up.

nwf, 2020-07-01:
Closed with #64, I think.

closed 2020-07-01 by nwf