riscv/sail-riscv #58: Sv39 virtual address msb check missing?

scottj97, 2020-06-08:
The priv spec (section 4.4.1 on "Addressing and Memory Protection") says:

> Instruction fetch addresses and load and store effective addresses, which are 64 bits, must have bits 63-39 all equal to bit 38, or else a page-fault exception will occur.

But I am seeing a case where the Sail model successfully performs a load, translating using bits [38:0] of the vaddr, and ignoring the upper bits. Spike (as expected) takes a load page fault.

pmundkur, 2020-06-09:
Good catch; I'd noticed this ages ago but forgotten to fix it.  Branch `fix_invalid_vaddr` should fix this, but it would be great if you could verify.

scottj97, 2020-06-09:
Tried it (Sv39 only) and it works great!

pmundkur, 2020-06-09:
great, thanks.

commit 6594b770 (2020-06-09): Properly handle invalid virtual addresses in address translation.
Fixes #58.
closed 2020-06-09 by pmundkur