riscv/sail-riscv #59: Cannot build sail-riscv

Imperas, 2020-06-15:
Hi All,
I am trying to run the riscv sail model against the riscv-compliance suite, but cannot build the simulator - I am on Ubuntu
I have installed ocaml
ocaml -version
The OCaml toplevel, version 4.06.1

I have sail - doesn't seem to take a version flag, but I get the following
sail -version
sail: unknown option '-version'.
Sail 0.10 (sail2 @ opam)
usage: sail <options> <file1.sail> ... <fileN.sail>

I have cloned the master branch - is there a release branch that should be used ?
git clone https://github.com/rems-project/sail-riscv.git

when I attempt to run make as per the instructions I get the following

ARCH=RV32 make
mkdir -p generated_definitions/ocaml/RV32
/home/moore/.opam/4.06.1/bin/sail -dno_cast -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV32 -o riscv model/prelude.sail model/prelude_mapping.sail model/riscv_xlen32.sail model/riscv_flen_F.sail model/prelude_mem_metadata.sail model/prelude_mem.sail model/riscv_types_common.sail model/riscv_types_ext.sail model/riscv_types.sail model/riscv_vmem_types.sail model/riscv_reg_type.sail model/riscv_freg_type.sail model/riscv_regs.sail model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_pmp_regs.sail model/riscv_pmp_control.sail model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_misa_ext.sail model/riscv_csr_map.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail model/riscv_next_control.sail model/riscv_softfloat_interface.sail model/riscv_fdext_regs.sail model/riscv_fdext_control.sail model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_pte.sail model/riscv_ptw.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv32.sail model/riscv_vmem_rv32.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail model/riscv_insts_next.sail model/riscv_insts_fext.sail model/riscv_insts_cfext.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail model/riscv_step.sail
Syntax error:
[model/riscv_insts_begin.sail]:15:30-30
15 |val encdec : ast <-> bits(32) effect {rreg}
   |                              ^
   | current token: effect
make: *** [Makefile:189: generated_definitions/ocaml/RV32/riscv.ml] Error 1

hmm, where did I do wrong ?

Thx
Lee

pmundkur, 2020-06-16:
You should be able to use the master branch of sail-riscv. It looks like you are using an outdated sail compiler (version 0.10). The current version in opam is 0.13.

Imperas, 2020-06-16:
@pmundkur 
Not sure what I did wrong - I followed the instructions and it gave me that version of sail - what should I do differently ?

Thx
Lee

PeterSewell, 2020-06-16:
perhaps you have an old opam install that needs its database updated?

Alasdair, 2020-06-16:
The opam package now on https://opam.ocaml.org/packages/sail/ should be 0.13. Make sure you have opam version >= 2.0. If you are using an older linux distro you may have an old version of opam that will not install newer packages. For such an old language the OCaml ecosystem sometimes moves fast and breaks things at an annoying rate. Pretty much everything has gone to opam 2 now, so while we supported opam 1 for quite a while using our own opam repository, it is no longer viable due to other packages switching to opam 2.

Our install guide was a little out of date. I have updated it now.

Alasdair, 2020-06-16:
You may also have to do an `opam update` to get the latest packages

Imperas, 2020-06-16:
My distro is Ubuntu 18
the version of opam is 1.2.2, so it is not greater than 2.0
looks like  I cannot install opam >1.22 on Ubuntu 18, apt search does not find a later version

What is the minimum OS in order to go through the build/install of sail-riscv?

bacam, 2020-06-16:
You can obtain a more recent version of opam that will install on Ubuntu 18.04 at http://opam.ocaml.org/doc/Install.html.

Imperas, 2020-06-17:
@bacam 
Thanks, I installed opam via your suggestion https://github.com/rems-project/sail-riscv/issues/59#issuecomment-644944273, and went back to stage 1 of the total install

I now have 
opam --version 
2.0.7

ocaml --version
The OCaml toplevel, version 4.06.1

I performed the additional apt-get 
sudo apt-get install build-essential libgmp-dev z3 pkg-config zlib1g-dev

I removed my entire .opam downloads (~/.opam, ~/.opam.1.2.2)
I then re-ran 
opam init

now when I perform 
opam install sail

so far so good - if I now try to build sail-riscv as per instructions
ARCH=RV32 make
I get a lot of warnings of the following form (which I guess I can ignore)
```
Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:76:7-12
76 |  then match n {
   |       ^---^
   | 
Most general matched pattern is 0

Then I have a failure 
```
mkdir -p ocaml_emulator/_sbuild
cp ocaml_emulator/_tags ocaml_emulator/platform.ml ocaml_emulator/platform_impl.ml ocaml_emulator/softfloat.ml ocaml_emulator/riscv_ocaml_sim.ml generated_definitions/ocaml/RV32/*.ml ocaml_emulator/_sbuild
cd ocaml_emulator/_sbuild && ocamlbuild -use-ocamlfind riscv_ocaml_sim.native
+ ocamlfind ocamldep -package lem -package linksem -package zarith -modules riscv_ocaml_sim.ml > riscv_ocaml_sim.ml.depends
ocamlfind: Package `lem' not found
Command exited with code 2.
Compilation unsuccessful after building 1 target (0 cached) in 00:00:00.
Makefile:192: recipe for target 'ocaml_emulator/_sbuild/riscv_ocaml_sim.native' failed
make: *** [ocaml_emulator/_sbuild/riscv_ocaml_sim.native] Error 10
```

```

Imperas, 2020-06-17:
Update, I think I have spotted the issue
I now have 2 ocaml versions installed
/usr/bin/ocaml
/home/moore/.opam/4.06.1/bin/ocaml

the latter is the correct one, this is selected by running 
eval $(opam config env)

but this is not persistent, it is only part of the shell environment
so a new shell reverts back to using /usr/bin/ocaml

running the eval command above then running make (or build_simulators.sh)
now seems to do the trick

I still see lots and lots of warnings - none of which I can determine if they are relevant or not

certainly not for the faint-hearted, finally can I suggest that the build_simulators.sh script tests the return code from the make of each target - otherwise it will not know that it has failed - I will try to issue a PR for this.

Thx
Lee

Alasdair, 2020-06-17:
Yes the sail pattern completeness check can give spurious warnings. It's a bit tricky because the type level constraints can restrict the patterns in ways that mean a purely syntactic check isn't enough. At the moment it is set up to always warns in certain bad cases (like a misspelled enum variant acting as a wildcard), at the expense of false positives

Imperas, 2020-06-18:
OK thanks, closing this issue

closed 2020-06-18 by Imperas