riscv/sail-riscv #67: compressed hints treated as illegal ops instead of nops

neelgala, 2020-08-29:
According to the isa manual the operation: ```c.add zero, ra``` is a hint operation and effectively should be treated as nop. However, SAIL seems to interpret this as an illegal instruction.

Following is the log using the C emulator:

```
mem[X,0x800001EC] -> 0x9006                                                                                
[168] [M]: 0x800001EC (0x9006) c.illegal 36870                                                             
trapping from M to M to handle illegal-instruction                                                         
handling exc#0x02 at priv M with tval 0x00000000                                                           
CSR mstatus <- 0x00001800  
```

Other hints like ```c.nop 0``` also are interpreted as illegal instructions.

neelgala, 2020-08-29:
A possible fix for c_add instruction would be to remove the ``` rsd != zreg``` condition in the following lines

https://github.com/rems-project/sail-riscv/blob/1bb74ef9664c63daee673770527d7c477f06288a/model/riscv_insts_cext.sail#L545-L556

However I am not completely sure if thats the only place that needs to be changed.

scottj97, 2020-08-29:
Dup of #31

neelgala, 2020-08-30:
yup.. did not see that before opening this. Since #31 is closed I am going to keep this issue open until the merge is performed.

commit c138a086 (2020-09-04): Handle hints explicitly in order to not trap on them.

This currently maps their assembly renditions to non-standard instructions to preserve bidirectional mappings.
Fixes #67 and #29.