riscv/sail-riscv #68: state monad in sail-riscv coq definition

ChrisYin, 2020-10-10:
Hi
This is a great job for ISA semantics!
I successfully compiled the coq model with the prompt_monad, which can be used for concurrent semantics.

1. I want to ask that has **the state monad** mentioned in your POPL 2019 paper been tested in risc-v model yet?

2. Another question is that how to define **the effect of instructions of monad**, like defining the effect of "Read_reg", "Write_reg", etc. Although it has a register accessor defined in riscv-type.v, it has been used yet.

bacam, 2020-10-16:
The state monad is defined in the same way for Coq that it is for Isabelle, and you can use it in the same way by lifting definitions that use the prompt monad into the state monad:
```
Require Import Sail.Base.
Require Import Sail.State_lifting.
Require Import riscv_types.
Require Import riscv.

Check (decode (mword_of_int 0)).
(*  Prompt monad
decode (mword_of_int 0)
     : M ast
*)
Check (liftState register_accessors (decode (mword_of_int 0))).
(*  State monad
liftState register_accessors (decode (mword_of_int 0))
     : State_monad.monadS regstate ast exception
*)
```

The register accessors are used by the lifting (they are passed as an argument to `liftState` in the example code above).

One of my colleagues is investigating a simpler setup for a state monad to integrate Sail ISA semantics with Iris separation logic system.  If this works out then we might provide direct support for it in Sail.

allenjbaum, 2021-09-29:
Is there any work required to be completed before closing this issue, or any unresolved question?
Otherwise, can we close this?

bacam, 2021-10-06:
I don't think there's anything else, unless @ChrisYin needs more detail?

closed 2021-11-02 by ChrisYin