riscv/sail-riscv #71: Support for when Xlen != Flen

abukharmeh, 2020-10-27:
Hello, I am reviewing the model before implementing parts of the proposed ZFinX extension, and I am wondering if there is a plan to support cases when Xlen != Flen ? 

Also is there any guidelines one should follow when planning to contribute to the model repo ? 

Thank you very much !

PeterSewell, 2020-10-27:
> Hello, I am reviewing the model before implementing parts of the proposed
> ZFinX extension, and I am wondering if there is a plan to support cases
> when Xlen != Flen ?
>
> Also is there any guidelines one should follow when planning to contribute
> to the model repo ?

Prashanth might have more to add, but there's a guide for extending the
model there already.

best,
Peter

martinberger, 2021-09-27:
I discussed this with Ibrahim, who thinks this needs to be handled for his upcoming PR on Zfinx. He has some ideas. So I'm reopening.