riscv/sail-riscv #72: Range check problem

soberl, 2020-10-29:
https://github.com/rems-project/sail-riscv/blob/ae6cb1de092e9ea727e2318d76e9b88999bbee59/model/riscv_pmp_control.sail#L60

I think it should be "if      (addr + width <_u lo) | (hi <=_u addr)" instead.
Suppose there are 2 ranges: [0x0, 0x200000) and [0x200000, 0x300000). Giving an address 0x200000 with any size, it should be belong to the 2nd range.

allenjbaum, 2021-02-26:
This appears to be correct - 
ranges for PMP NAPOT format define regions at base..base+regionsize-1
ranges for PMP range    format define regions at base_lo..base_hi-1
This is a bug in the RISC-V Sail model

scottj97, 2021-02-26:
It's broken on the low end, too. With your example ranges, consider an access to addr=0x1ffffc width=4. This is outside of the second range, yet `(addr + width <_u lo)` would incorrectly say that it's within that entry.

Correct:
```
                        if      (addr + width <=_u lo) | (hi <=_u addr)
```

soberl, 2021-02-26:
Yes.
For easier understanding, maybe (assert width > 0):
`if      (addr >= _u hi) | (addr + width <=_u lo) `

scottj97, 2021-02-27:
I wonder if there's similar corner cases here when accessing the largest-addressed bytes in memory. addr=0xFFFFFFFC width=4, so addr+width wraps around.

Since addr is of type `xlenbits`, this would apply only to RV32, since RV64 physical addresses can't be so large as to wrap. (Right?)

soberl, 2021-05-06:
See the fix as commented.

scottj97, 2021-05-06:
@soberl what fix? What comment? #85 is still open?

soberl, 2021-05-06:
Ah, yes, it's still open. Let me reopen this issue.

> @soberl what fix? What comment? #85 is still open?

rmn30, 2021-07-23:
I believe this is fixed by https://github.com/rems-project/sail-riscv/pull/85 but someone should check.

scottj97, 2021-07-23:
I can confirm it is fixed.

closed 2021-07-23 by rmn30