riscv/sail-riscv #98: RVFI-supporting emulator no longer works with ELF test files

rbanerjee20, 2021-07-19:
Hello,

I was working with the C emulator that supports the RVFI-DII format (as generated for TestRIG in its [Makefile](https://github.com/CTSRD-CHERI/TestRIG/blob/e0f1d191deb2b15cf8b86ed872031b472f0c311e/Makefile#L163)), and I noticed that it no longer works the same way with the provided ELF test files in `test/riscv-tests`.

For example, if I pass [`rv32ui-p-add.elf`](https://github.com/rems-project/sail-riscv/blob/master/test/riscv-tests/rv32ui-p-add.elf) to the standard C emulator (just produced by running `ARCH=32 make`), the test passes. However, if I pass this same test file to the emulator extended with support for RVFI instruction streams, I get infinite faults of the form:

```bash
...

[7907] [M]: 0x00000000 (0x0000) c.illegal 0
trapping from M to M to handle illegal-instruction
handling exc#0x02 at priv M with tval 0x00000000
CSR mstatus <- 0x00001800

[7908] [M]: 0x00000000 (0x0000) c.illegal 0
trapping from M to M to handle illegal-instruction
handling exc#0x02 at priv M with tval 0x00000000
CSR mstatus <- 0x00001800

...
```

I was wondering whether the 'extended' emulator is only designed to deal with RVFI streams, or whether it is supposed to be able to deal with ELF files as well, just like the standard one.

jrtc27, 2021-09-04:
I'm surprised that ever even worked. I don't think we've ever intended for something like that to be supported, the current implementation has the RVFI and DII parts tightly coupled.

allenjbaum, 2021-09-29:
Could someone explain what "extended" is in this context? 
And I'm a bit puzzled about the RVFI being tightly coupled vs. what was attempted.
Is it the difference of using RVFI and RVFI-DII, and that Sail only intended to support the latter?
IF so, can we close this issue with a note that says that (And possibly document in the Sail docs)

jrtc27, 2021-09-29:
> Could someone explain what "extended" is in this context?

"Extended" is the wrong word, as some parts of the model are _replaced_, you can't just take c_emulator/riscv_rvfi_RV64 and try and use it as if it were c_emulator/riscv_sim_RV64. (Though in theory I believe you could make a unified binary with a bit of work, just not something we cared about.)

> And I'm a bit puzzled about the RVFI being tightly coupled vs. what was attempted.
> Is it the difference of using RVFI and RVFI-DII, and that Sail only intended to support the latter?
> IF so, can we close this issue with a note that says that (And possibly document in the Sail docs)

Yes, your understanding is correct. The README currently says:

> For tandem verification of random instruction streams, the tools support the protocols used in TestRIG to directly inject instructions into the C emulator and produce trace information in RVFI format.

I guess the use of "and" is a little ambiguous; I read it as "support (A and B)" but you could parse it as "(support A) and (support B)".

closed 2021-09-29 by jrtc27