riscv/sail-riscv #100: Cannot build sail-riscv

OYounis, 2021-08-02:
Hello all, 
It seems that I cannot build Sail-RISCV.

`make: /home/oyounis/.opam/default/lib/sail/sail: Command not found`
`make: *** [Makefile:191: generated_definitions/ocaml/RV32/riscv.ml] Error 127`

I guess the problem is that I cannot set the `SAIL_DIR` variable to the right directory:

 `export SAIL_DIR=/home/oyounis/.opam/default/lib/sail`
`ARCH=RV32 make`

rmn30, 2021-08-02:
Hi,

What is the output of `echo $PATH`? Did you run `eval $(opam config env)` in the current shell to configure the environment?

rmn30, 2021-08-02:
The sail binary is installed in `$OPAM_ROOT/bin/sail` If you `unset SAIL_DIR` the Makefile should attempt to get the correct values: https://github.com/rems-project/sail-riscv/blob/master/Makefile#L99

rmn30, 2021-08-02:
I think the confusion here is because of this line in the sail-riscv README:
> Install Sail, either via `opam` OR by building Sail from source and setting `SAIL_DIR` in your environment pointing to its top-level directory.

Note that I have added emphasis on the 'or' because this is intented as alternative options for two different scenarios:
1. Most people can just use `sail` installed from `opam` and do not need to set `SAIL_DIR`.
2. If for some reason the `opam` packaged version of `sail` does not work for you can set `SAIL_DIR` to a local checkout of the `sail` source code repo (e.g. you need the very latest development version of `sail` for a bug fix or new feature that is not in the `opam` version or you are actively working on sail)

We should update the README to make this clearer.

OYounis, 2021-08-02:
Thank you for the clarification. The make process is now working but it seems to be stuck.
I do not know if this is normal or not but it has been like this for approximately 2 hours.

[image]

rmn30, 2021-08-03:
That output looks normal but it should not take that long to compile. It might be a problem with the SMT solver (z3) used for type checking. What is the output of `z3 --version`? What platform are you running on?

OYounis, 2021-08-03:
`Z3 version 4.8.7 - 64 bit`
I am running on WSL Ubuntu. I will try out the z3 recent release.

rmn30, 2021-08-03:
I've updated the README with this commit cf099be2a1aff1f326dc855875d5732607e28752 . Please confirm if you get it building and I will close this issue.

closed 2021-08-03 by rmn30