riscv/sail-riscv #102: Random test generation

OYounis, 2021-08-06:
Hello all, 
According to the Readme, Sail can be used for random test generation. But I cannot find how to do so.

jrtc27, 2021-09-04:
That uses https://github.com/rems-project/sail-riscv-test-generation

OYounis, 2021-09-05:
Is there any documentation on how to use it?

bacam, 2021-09-05:
There isn't much, although I did add a bit to the Sail manual about the underlying feature.  This is a pretty simple form of test generation, though: it just produces random instructions from the model for use with [the TestRIG tool](https://github.com/CTSRD-CHERI/TestRIG/).  We have used Sail to produce more self-contained random tests for another architecture using an accompanying symbolic execution tool, but not currently for RISC-V.

allenjbaum, 2021-09-29:
IF this issue is resolved, can we close it? Or does this require some documentation to be written?

martinberger, 2021-10-02:
This is not a question for the Golden Model repo, but a Sail compiler question, so I suggest to discuss at the [Sail Github Issue page](https://github.com/rems-project/sail/issues).

closed 2021-10-02 by martinberger