riscv/sail-riscv #104: sail-riscv build fails

vignajeth, 2021-08-13:
Hi, 
   I am trying to install sail-riscv. I installed sail using opam, but i get the following error when i run `make` inside sail-riscv folder.
```
[user@machine sail-riscv]$ make all                                                                                          mkdir -p generated_definitions/ocaml/RV64
sail -dno_cast -ocaml -ocaml-nobuild -ocaml_build_dir generated_definitions/ocaml/RV64 -o riscv model/prelude.sail
model/prelude_mapping.sail model/riscv_xlen64.sail model/riscv_flen_D.sail model/prelude_mem_metadata.sail
model/prelude_mem.sail model/riscv_types_common.sail model/riscv_types_ext.sail model/riscv_types.sail
model/riscv_vmem_types.sail model/riscv_reg_type.sail model/riscv_freg_type.sail model/riscv_regs.sail
model/riscv_pc_access.sail model/riscv_sys_regs.sail model/riscv_pmp_regs.sail model/riscv_pmp_control.sail
model/riscv_ext_regs.sail model/riscv_addr_checks_common.sail model/riscv_addr_checks.sail model/riscv_misa_ext.sail
model/riscv_csr_map.sail model/riscv_next_regs.sail model/riscv_sys_exceptions.sail model/riscv_sync_exception.sail
model/riscv_next_control.sail model/riscv_softfloat_interface.sail model/riscv_fdext_regs.sail model/riscv_fdext_control.sail
model/riscv_csr_ext.sail model/riscv_sys_control.sail model/riscv_platform.sail model/riscv_mem.sail model/riscv_pte.sail
model/riscv_ptw.sail model/riscv_vmem_common.sail model/riscv_vmem_tlb.sail model/riscv_vmem_sv39.sail
model/riscv_vmem_sv48.sail model/riscv_vmem_rv64.sail model/riscv_insts_begin.sail model/riscv_insts_base.sail
model/riscv_insts_aext.sail model/riscv_insts_cext.sail model/riscv_insts_mext.sail model/riscv_insts_zicsr.sail
model/riscv_insts_next.sail model/riscv_insts_hints.sail model/riscv_insts_fext.sail model/riscv_insts_cfext.sail
model/riscv_insts_dext.sail model/riscv_insts_cdext.sail model/riscv_jalr_seq.sail model/riscv_insts_end.sail
model/riscv_step_common.sail model/riscv_step_ext.sail model/riscv_decode_ext.sail model/riscv_fetch.sail
model/riscv_step.sail model/riscv_analysis.sail
Error:
Error when calling smt: Reporting.Fatal_error(_)
make: *** [generated_definitions/ocaml/RV64/riscv.ml] Error 1
```

The version of sail and opam i use is
```
[user@machine sail-riscv]$ opam --version
2.0.8
[user@machine sail-riscv]$ sail -v
Sail 0.14 (sail2 @ opam)
```
i have 'home_dir/.opam/ocaml-base-compiler.4.06.1/bin' in my PATH and OPAM_SWITCH_PREFIX, CAML_LD_LIBRARY_PATH, OCAML_TOPLEVEL_PATH, PKG_CONFIG_PATH, MANPATH related to opam are configured

It would be great if you can help me here :smiley:

Alasdair, 2021-08-13:
What is your z3 version? I can try to reproduce with the same version

allenjbaum, 2021-09-29:
If there is no response byOct 13, can we close this issue as operator error, please?

closed 2021-10-01 by vignajeth