riscv/sail-riscv #106: Lem and adding the plat_get_16_random_bits entropy source function

ben-marshall, 2021-09-08:
Hi there

I'm investigating why my PR (#99 ) is failing and this is the error message I get locally:

```
work@ben-TF:~/riscv/riscv-crypto/extern/sail-riscv$ make ARCH=RV64
mkdir -p ocaml_emulator/_sbuild
cp ocaml_emulator/_tags ocaml_emulator/platform.ml ocaml_emulator/platform_impl.ml ocaml_emulator/softfloat.ml ocaml_emulator/riscv_ocaml_sim.ml generated_definitions/ocaml/RV64/*.ml ocaml_emulator/_sbuild
cd ocaml_emulator/_sbuild && ocamlbuild -use-ocamlfind riscv_ocaml_sim.native
Finished, 25 targets (25 cached) in 00:00:00.
lem -isa -outdir generated_definitions/isabelle/RV64 -lib Sail=/home/work/.opam/ocaml-base-compiler.4.06.1/share/sail/src/lem_interp -lib Sail=/home/work/.opam/ocaml-base-compiler.4.06.1/share/sail/src/gen_lib \
	handwritten_support/0.11/riscv_extras.lem handwritten_support/0.11/mem_metadata.lem handwritten_support/0.11/riscv_extras_fdext.lem \
	generated_definitions/lem/RV64/riscv_types.lem \
	generated_definitions/lem/RV64/riscv.lem
File "generated_definitions/lem/RV64/riscv.lem", line 14941, character 24 to line 14941, character 46
  Type error: unbound variable: plat_get_16_random_bits
make: *** [Makefile:318: generated_definitions/isabelle/RV64/Riscv.thy] Error 1
```

I am responsible for adding the `plat_get_16_random_bits` function. This was recommended to me by the Sail authors, since the entropy source model "calls out" to the execution platform to get entropy. However, I have no idea what the "lem" tool is, or why it should complain about my function.  

I can build the rv32/64 c and ocaml models fine with
```
make ARCH=RV32 csim osim
make ARCH=RV64 csim osim
```

But running `make all` for either architecture causes this problem.

So, I know I've broken _something_, but I'm not enough of a Sail expert to know exactly what.

Thanks,
Ben

bauereiss, 2021-09-09:
Lem is a tool and language that is used as an intermediate target for translating Sail specifications to the Isabelle and HOL4 theorem provers.  The error is due to the fact that your function is declared as an external function for the `lem` backend, but no implementation is provided.  You could do that either in the corresponding Sail file (i.e. providing a default Sail implementation and removing the `lem` external binding from the `val` spec;  Sail will still use any external bindings declared for other backends) or in a Lem file.  There seems to be at least one extension following the latter approach in the `handwritten_support/0.11/riscv_extras_fdext.lem` file (although all those definitions seem to be stubs that don't actually do anything).

ben-marshall, 2021-09-10:
Thanks for explaining this, I'll have a go at adding the relevant bits and see how I get on.

Cheers,
Ben

ben-marshall, 2021-09-15:
Right, this solved it for me, thank you for your help! :)

closed 2021-09-15 by ben-marshall