riscv/sail-riscv #109: Risc-V Sail code has no version string

allenjbaum, 2021-10-01:
We would like to implement a versioning string so that we can easily tell if a version with some non-backwards compatible change is being used by checking the version. I don't think it matters what the format of the version is - it just need to be monotonically increasing, and not a random hash value. The riscof team has implemented this as a simple CI script, I believe, and you might be able to follow their recipe.

neelgala, 2021-10-01:
riscof uses semantic versioning. The CI script basically picks up the [latest version](https://github.com/riscv-software-src/riscof/blob/master/.github/workflows/main.yml#L26-L29) present in the changelog.md and creates a release for the same version. 

If every PR should create a release then every PR will need to include an entry in the CHANGELOG.md.

We also have a definition of how the [versioning number](https://github.com/riscv-software-src/riscof/blob/master/CONTRIBUTING.rst#versioning) must be assigned.

jrtc27, 2021-10-01:
Is semantic versioning really needed though? The architecture should always be backwards-compatible.

neelgala, 2021-10-01:
not really.. any monotonically increasing versioning system will work. If you are using the CI from riscof then you simply need to change the format of the string inside `[]` each release entry of the changelog.md file. Calendar versioning will also work.

neelgala, 2021-10-01:
its not just for comparison..  the compliance/compatibility reports (html-base) will require mentioning what version of sail was used to reference signature generations. Having something that we can quickly infer as old and new would be best.

allenjbaum, 2021-10-01:
If counting the number of commits is easy, that's fine. Semantic versioning
gives a bit more information, but not enough to require it if its any more
work.

In the particular case that started all this, it wasn't the
architecture that was changing, it was the reporting that was an issue.
Specifically: issue 16, where CSRwrites to restricted view CSRs (e.g.
xSTATUS) always report the full CSR (MSTATUS) in Spike rather than the
actual CSR being written (SStatus or UStatus) that Sail reports, and the
request was to make it match Spike.
It's probably more useful to do it the Spike way, since SW that depends on
it can do the filtering it needs, but couldn't do it the other way around.

That is an incompatible change that other SW (which use that reporting
information) would notice - but isn't an architectural change.
Having said that: there are architectural changes that are not backwards
compatible,
but they only get made if its highly unlikely anyone actually depends on
that behavior.
You can look at the priv spec preface that mentions them.

martinberger, 2021-10-02:
> We would like to implement a versioning string 

Hello Allen, where do you want the version strings to be, in each .sail file, or once, somewhere in the top-level directory of the master branch?

allenjbaum, 2021-10-03:
Top level, master branch is fine; that's the one that vendors will be using
(and/or will be in the docker image)

martinberger, 2021-10-03:
How about we have a top-level file `VERSION.md` in `master` that contains as first line the git tag for the release?

jrtc27, 2021-10-03:
Markdown isn't helpful, it requires you to look at the source. Better to have something embedded in the binary itself. Perhaps the output of `git describe` (possibly with `--first-parent`, depending on how you want to think of merge commits)? If you have a tag checked out, it prints the name of that tag, otherwise it prints the name of the most recent tag on your branch, plus the number of commits made since then, plus the current hash. So, for example, master is currently `0.5-104-g9f71c75`, meaning commit 9f71c75 (how many characters gets printed is configurable), which is 104 commits newer than 0.5 (including the "other" sides of merges), or `0.5-70-g9f71c75` if you pass `--first-parent` to consider merges as a single commit.

martinberger, 2021-10-03:
>  something embedded in the binary itself. 

Which binaries have you got in mind?

jrtc27, 2021-10-03:
> > something embedded in the binary itself.
> 
> Which binaries have you got in mind?

riscv_sim_RV64 etc

martinberger, 2021-10-03:
How do you suggest that version be extracted from binaries, by grepping or something like `--version`?

jrtc27, 2021-10-03:
> How do you suggest that version be extracted from binaries, by grepping or something like a `--version` command line switch?

The latter was what I assumed would be done, like any other versioned tool

martinberger, 2021-10-03:
That's fine. Does Sail offer a reliable facility to drop this kind of information into all it's backends (OCaml, C, HOL ...) or would this have to be done on a per-backend basis?

jrtc27, 2021-10-03:
No, command-line parsing is done in each platform's wrapper (riscv_sim.c and riscv_ocaml_sim.ml), though in theory you could make it a Sail string and have each platform read that so there's at least only one copy of the version, but that might be a bit unnecessarily complicated (and might require flags to stop Sail optimising out the variables).

The prover snapshots don't support any notion of command-line arguments, nor do you "run" them, but their README.md's already automatically document what commit they were built from, so could be modified to match the C and OCaml versioning scheme.

martinberger, 2021-10-04:
@jrtc27  Sounds all very reasonable, I'm happy to have such an addition. Now we 'only' need to agree on the exact naming scheme. I have no particular preferences. It sounds like @allenjbaum and @neelgala are OK with something like `0.5-104-g9f71c75`, do I understand this correctly?

jordancarlin, 2025-01-14:
@edolnx this might be something to consider as you are looking at versioned releases of the model

jordancarlin, 2025-07-15:
Added in #887.