riscv/sail-riscv #111: riscv-arch-test signature file generation size option

MarcKarasek, 2021-10-04:
Would like to have an option to dump the signature file used by the riscv-arch-test (riscof) in either 32 bit values or 64 bit values.

This is so the barrier of entry for using the riscv-arch-test suite is easier.  Currently if you are on a RV64 dut, then the signature is dumped out as 64 bit values (writememh()) and the resulting signature needs to be post processed into 32 bit values to match what sail is generating.

jrtc27, 2021-10-04:
What signature file? There is no riscof integration in Sail, just a PR whose general approach I have issues with.

neelgala, 2021-10-04:
The C-simulator of Sail uses the --test-signature cli to dump out a memory region of the test encapsulated between begin_signature and eng_signature labels. This is used here : https://github.com/riscv-non-isa/riscv-arch-test/blob/master/riscv-target/sail-riscv-c/device/rv32i_m/I/Makefile.include#L9

neelgala, 2021-10-04:
spike has this option of `+signature-granularity=<num-bytes-per-line>` which allows the user to specify the number of bytes to be dumped per line of the signature file. Something similar if available in SAIL would be useful.

jrtc27, 2021-10-04:
Oh ok that thing (which also exists for the OCaml simulator, I might add). Making that configurable isn't hard.

MarcKarasek, 2021-10-04:
At the end of a test case run in sail it generates a signature file (dump
of memory) to the system.

I was told this was from sail.

MarcKarasek, 2021-10-04:
It is this option :
riscv_sim_RV64 --test-signature=<filename> <elf file>

This is what is run for a test case...

MarcKarasek, 2021-10-04:
Great htat would help us out a lot in how the signatures are generated
between RV32 and RV64 bit DUTs..

abukharmeh, 2022-03-08:
@MarcKarasek  This has landed in the master branch, so I think we can close this issue?

MarcKarasek, 2022-03-08:
ok I have been tied up with other issues ..  But am now getting back to
running riscof on our DUT..

jordancarlin, 2024-12-05:
Seems like this was finished a while ago with #136 and should be closed.

closed 2024-12-05 by Timmmm