riscv/sail-riscv #117: TOR pmp matching with same hi and lo pmpaddr

stuarth-sfu, 2021-10-16:
In the case the ranges for a PMP check in TOR mode are the same (ie lo == hi), I am not sure the Sail specification/model will do the right thing.
I believe the correct behaviour here will always be to not match the entry, however the Sail specification would give a partial match.
The effects of this are subtle, because a no match will then go on to check the next entry, wheras a part match will cause the whole check to fail.

Take the following example :
if  pmpaddr[0] == pmpaddr[1] == 0x...10000
the address to check is 0x....FFFF for a 2 byte access
This means the two bytes to check are 0x....FFFF and 0x...10000 neither of these should match according to The privileged architecture (section 3.7.1, 20210921-draft), so the result should be "no match" for that entry

here lo == hi == 0x...10000, addr = 0x....FFFF, addr + width = 0x...10001
In the sail spec this will result in a partial match because the first check of the range values is for > not >=, then the second if condition, (addr + width <=_u lo) | (hi <=_u addr), will not be so it falls into the second condition, (lo <=_u addr) & (addr + width <=_u hi), which is also not met so results in a partial match.

Should the initial misconfiguration check,  hi <_u lo, be checking for hi <=_u lo?

allenjbaum, 2022-01-12:
This is a confirmed  Sail bug. A partial match is if the first or last byte is entirely within a range, and the other isn't.
if the low and high range limits are the same, then the range size is zero, and it is impossible for either end to be inside the range, because there is no inside....'
I don't think I can tag this as a bug, but it should be.

ptomsich, 2023-05-29:
@allenjbaum Allen, can you confirm this has been addressed by PR #141 ?

allenjbaum, 2023-06-01:
Given my comments- I think so, but it’s been quite a while…

closed 2024-05-07 by Timmmm