riscv/sail-riscv #124: Feature request: Sail code formatter for our CI

martinberger, 2021-11-05:
I've made a request for a Sail auto-code formatter [1] on Sail's GitHub. Given that they are an academic group, this is unlikely to be their highest priority. But it would be rather useful for RISC-V. Maybe such a formatter already exists in one of the many commercial teams that use RISC-V and they might be interested in donating it?

1. https://github.com/rems-project/sail/issues/158

trdthg, 2025-06-10:
Considering the current state of sail-riscv, it is almost impossible to have a complete and sufficiently detailed formatting rules.

for example:

- for '^'
  we can choose to have all elements wrap to a new line, or to have them wrap automatically based on line_width, but neither is currently followed in the code.
  ```

  mapping clause assembly = F_BIN_F_TYPE_D(rs2, rs1, rd, op)
                        <-> f_bin_f_type_mnemonic_D(op)
                            ^ spc() ^ freg_or_reg_name(rd)
                            ^ sep() ^ freg_or_reg_name(rs1)
                            ^ sep() ^ freg_or_reg_name(rs2)
  ``` 

  Similar examples also include vector concat syntax. `bits1 @ bits2`

- for Index
  
  This example would be difficult to implement, especially when you consider that the index itself can be any complex expression.

  ```
  val zvk_sm4_subword : bits(32) -> bits(32)
  function zvk_sm4_subword(x) = {
    sm4_sbox(x[31..24]) @
    sm4_sbox(x[23..16]) @
    sm4_sbox(x[15.. 8]) @
    sm4_sbox(x[ 7.. 0])
  }
  ```

- for List
  For crypt extensions, they have a large number of elements, but the data itself usually is just bit vector literals. Automatic wrap seems to be a good choice.  
  ```
  let sm4_sbox_table : vector(256, bits(8)) = [
  ..., ...
  ]
  ```

  But for other lists, have all elements wrap might be the better option.

For these complex examples above, I can't think of a good way to handle them unless you impose strict limitations in the rules. For instance, when trying to align `let foo = bar` statments, we could have limit that only if bar is an Int.

But for some simpler cases, like adding a trailing comma after the last mapping element, these can be done immediately.

I'm wondering how we can quickly apply these simple formatting rules to sail-model. Perhaps we need to add many Sail configuration options for various keywords, and disable them by default.

trdthg, 2025-06-10:
Not yet, but maybe better than a lot options. Sail attribute syntax seems a more suitable option `$[format(off)]`. I'll try when I get the chance.

Alasdair, 2025-06-10:
A format off attribute should be easy enough to add, there is already code for handling definitions it doesn't know how to format by just skipping them.