riscv/sail-riscv #138: Role of the Sail OCaml simulator in the golden model

martinberger, 2021-12-22:
Currently Sail can generate a C simulator (using the `-c` option) and an OCaml one (using `-ocaml`). The latter is much more slow. Keeping them in sync occasionally causes problems, in particular in the recent P-extension PR, where a hard-coded limitation in the OCaml compiler lead to trouble with the OCaml emulator, see [1]. There is currently also a small discrepancy in typing depending on whether C or OCaml are targeted [2].  While those problems can be fixed, I wonder about the use-cases for the OCaml simulator. I do understand the benefits of n-version programming, especially in the evolution of Sail itself. But how important is it for RISCV?

[1] https://github.com/riscv/sail-riscv/pull/132#issuecomment-987769583

[2] https://github.com/rems-project/sail/issues/122

b224hisl, 2022-02-10:
I have checked  rems-project/sail#122. I have meet the same problem. So could I just use -no_effects when building the ocaml simulator so that my code can pass the CI? I can make cim successfully.

jrtc27, 2022-02-10:
I've gone and filed https://github.com/rems-project/sail/issues/165 against Sail to try and get this fixed

martinberger, 2022-02-10:
> I've gone and filed 

Thanks. I didn't realise this was a regression. Reminds me that I should write more regression tests ...

jrtc27, 2022-02-10:
Well, Sail 1 and Sail 2 are similar but different languages, so more like saying it's a regression from Python 2 to Python 3. I'm pretty sure the RISC-V Sail model has always been for Sail 2, but wasn't involved that far back to know.

b224hisl, 2022-02-12:
> @b224hisl I think you are hitting hard-coded limitation in the OCaml compiler I mention above. Have you branched your code off from the proposed P-extension code [1]. Or are you adding a large number of OCaml ADT-constructors? (Sorry I'm in a rush, so no time to look into your code)
> 
> [1] Sail Model for RISC-V P Extension #132

I think I didn't invole any P-extension code. I do add a large number of items like `enum` because the number of v-instructions is a little large. Please let me know if there is anything you want me to hange

martinberger, 2022-02-12:
>  I do add a large number of items like `enum` because the number of v-instructions is a little large. 
@b224hisl This should be fixed now, see https://github.com/rems-project/sail/issues/165 Maybe pull the latest version and try locally. I think https://github.com/riscv/sail-riscv/blob/master/.github/workflows/compile.yml will automatically use the latest version of Sail if Cambridge have already published this fix to opam.

jordancarlin, 2024-10-12:
Can be closed now that #544 is merged

closed 2024-10-12 by Alasdair