riscv/sail-riscv #144: Ext_FetchAddr_Error is never used

Trolldemorted, 2022-01-23:
`Ext_FetchAddr_Check` declares an error variant:
```
union Ext_FetchAddr_Check ('a : Type) = {
  Ext_FetchAddr_OK  : xlenbits,  /* PC value to use for the actual fetch */
  Ext_FetchAddr_Error : 'a
}
```
but I cannot find any active usage of it, because `ext_fetch_check_pc` always returns the ok variant:
```
function ext_fetch_check_pc(start_pc : xlenbits, pc : xlenbits) -> Ext_FetchAddr_Check(ext_fetch_addr_error) =
  Ext_FetchAddr_OK(pc)
```

Should the error variant be used, or could it be removed?

jrtc27, 2022-01-23:
No; the Ext_/ext_ prefix means it's a hook for out-of-tree extensions to override, with a no-op default implementation provided for the standard model.

closed 2022-01-23 by jrtc27

Trolldemorted, 2022-01-23:
Thanks for the quick response!

So it is boilerplate code where specific implementations should fill in their instruction fetch check logic, and the RISC-V standard allows implementations to unaccept arbitrary fetch addresses?

jrtc27, 2022-01-23:
The RISC-V spec does not say anything about it because it does not know what non-standard or not-yet-ratified to-be-standard extensions do. Extensions can introduce arbitrary changes anywhere in the spec provided any backwards-incompatible ones are done in an opt-in manner. The Ext_/ext_ hooks present in sail-riscv are those that we have needed for implementing our CHERI-RISC-V extension, but could be used for other purposes too (and more added if other consumers appear that need hooks elsewhere).