riscv/sail-riscv #145: missing copyright headers

tzwaenn, 2022-01-27:
It seems that some of the recently added files do not have the standard copyright header that the reset of the repo has. I see it missing in
riscv_insts_zba.sail
riscv_insts_zbb.sail
riscv_insts_zbc.sail
riscv_insts_zbkb.sail
riscv_insts_zbkx.sail
riscv_insts_zbs.sail
riscv_insts_zfh.sail
riscv_insts_zkn.sail
riscv_insts_zks.sail
riscv_types_kext.sail

bilalsakhawat-10xe, 2022-02-02:
@tzwaenn riscv_insts_zfh.sail has the copyright header.
@jrtc27 can you please guide on this issue.

tzwaenn, 2022-02-02:
@bilalsakhawat-10xe That file has some copyright remarks. All other files seem to have an identical header with all the details. I assumed all files should have that header.

jrtc27, 2022-02-02:
Zb* should have Bilal's copyright
Zfh should have a copy of F's copyright plus Bilal's
Zk*/Kext should be Ben Marshall's

tzwaenn, 2022-02-02:
Great. There is not only copyright, but the BSD license info that is currently repeated in all files but the ones I listed.

jrtc27, 2022-02-02:
Yes, I know, clearly they're all implicitly licensed under the same BSD-2-Clause license as the rest of the model

tzwaenn, 2022-02-02:
Sure. It is just that legally, things are a lot easier if that info is repeated for each file...

jrtc27, 2022-02-02:
I realise; it should be fixed, but it doesn't require any great insight to know what it should be, unlike who the copyright holders are

tzwaenn, 2022-02-02:
Agreed. However, it still blocks re-distribution of the file for some legal departments if no license info is in the file (even if it could be inferred from the rest of the repo). Seems the headers were added here: aff2e7aed6fd946f40a223e8cc45c0db3608f9a6

jrtc27, 2022-02-02:
I don't deny that. That's why I said it should be fixed, and why the issue is still open. You asked for my advice (which you've now edited out) and so I just told you what's needed.

tzwaenn, 2023-05-26:
This would be addressed by the pending pull request #266.

ptomsich, 2023-05-29:
#266 has been applied.

Two requests:
- Someone with knowledge of the history needs to go back and check whether we have to update the LICENCE file with additional copyright holders and rerun apply_headers?
- We need a discussion at one of the next meetings on whether we really want to copy the entire header into every file (or just point back to LICENCE/COPYING/...) and if we should establish a CLA process before accepting PRs.

closed 2023-06-01 by tzwaenn