riscv/sail-riscv #148: Commandline options or parsing yaml

allenjbaum, 2022-02-02:
Feature request: There are many architectural options in RISC-V.  Many of them cannot be derived from examining processor state (including CSRs and eventually config ), but must be tested ( try something, see what answer you get). Entering them as individual command line options is unwieldiy, considering their number.  
A better option is to have Sail read and parse the risc-config format YAML file to assign variable values used to configure the csim, . Spike already has this option, though from a device-tree formatted file.
Another approach is to write a program that reads the YAML and produces the command line option string. This may b a better option because we would then have access to open source YAML parsers, and we can then more easily examine the options produced.

This approach has been used to create the Sail code that reads and writes CSRs (as opposed to producing commandline strings), This is being reconsidered to instead provide the parameters to a Sail CSR access function instead, which could . This could easily be extended to cover other variables, e.g. PM granularity, the number of PMP entries supported, unaligned access support, partial unaligned store support, trap cause proiorities, and many, many others.

abukharmeh, 2022-02-02:
Or maybe we can make CSIM read the YAML files and extern the needed values directly to SAIL instead of passing them as command line args ?! I

jrtc27, 2022-02-02:
That's what the first option says

abukharmeh, 2022-02-02:
`Sail read and parse`, I proposed to make the C simulator read the YAML file instead of SAIL code.

jrtc27, 2022-02-02:
Nobody's suggesting you write the parser in Sail, as a language it'd be terrible for writing that kind of code. The C parts of the C emulator are still part of the overall Sail compiled model, so it'd be just like how we use the soft-float C library rather than write that in Sail. The wording could perhaps have been clearer but I can guarantee that Allen did not mean what you think he did.

abukharmeh, 2022-02-02:
This made me think that Allen meant to say that the actual SAIL code parses YAML in his first option : `Another approach is to write a program that reads the YAML and produces the command line option string. This may b a better option because we would then have access to open source YAML parsers.` Otherwise he would not have emphasised on it in the second option ?

Anyways, if that's the first option, then I vote for that  !

allenjbaum, 2022-02-03:
I actually hadn't considered inserting the parser into the csim, to be frank, as opposed to a separate program that would generate the command line arguments that would be compiled in as constant values. (as opposed to trying to write in Sail, which sounds...hard).  So, writing it in Sail may have been the first thing I mentioned, but it certainly wasn't my first preference.
I think a separate program offers several advantages - you can easily see the commandline arguments that Sail will use to generate the Csim, it will be easier to debug, and the Csim is likely to run faster and be smaller because the options will be compile time constants instead of runtime variables. I see no downside to that approach.

I'm hoping that Sail is at least flexible enough to handle parsing the description of CSR WARL field mapping (after pre-parsing using this approach)

martinberger, 2022-02-03:
I agree that Sail should not parse YAML.  

Isn't the natural approach to have a top-level script that parses YAML and then configures and executes the building of the C simulator from Sail (this might require linking in additional libraries, when compiling the generated C, where required)?

abukharmeh, 2022-02-03:
I think that the cleanest solution would be CSIM parses YAML file, and expose some sort of structure to the compiled Sail code. 

Having a top level script that parses YAML and does something with it is not far from what #128 does AFAIK, though there it appears to generate SAIL code for the configuration (I think it could be argued that this is less deterministic).

I am not sure about passing configuration as command line argument, because that would expose an external interface that could/would be changed and people will cling to, the model at the moment initalise all global variables before parsing command line arguments, and that means we could not use cmd line args to pass all options without changing the way the model is initalised, and also I am not sure how far we can extend command line args until it becomes too much.

martinberger, 2022-02-03:
I wonder if one could not also argue that the Golden Model code is not, and should not be, concerned with implementation details of testing (such as: how to configure the extensions of interest?), as long as the model does not prevent any specific configuration. The purpose of the model is to express, as clearly and simply as possible, the ISA semantics of RISC-V.

As RISC-V becomes more popular we will get more and more requests to augment the Golden Model with this / that / the-other feature, which will all be jolly useful to somebody, but at the same time compromise the simplicity, readability and verifiability of the model. In particular, I expect to see requests for tracing all manner of model internal state during execution, so as to make it easier to compare the state of the Golden Model execution in a step-wise manner with RTL implementations.  I expect the Golden Model to evolve over time, to accommodate such requests, but I think a systematic approach is required, and needs some thought.  In the mean time, maybe we should be conservative and try and augment the model from the 'outside'.

allenjbaum, 2022-02-03:
The way that the csim is currently configured is to pass command-line
arguments to Sail, and Sail compiles the csim appropriately.
I am suggesting that we don't change that. What we are changing is how our
framework gets the configuration and formats it so that it can be passed on
the command line.
That separation seems to me to be cleaner, and more easily debugged, than
embedding it.
We could, of course, change that and embed it inside the Sail csim - but
what would the advantage of that be?

I'm also suggesting that #128
<https://github.com/riscv/sail-riscv/pull/128> does same
thing (if that is doable; I think it is more problematic than other
architectural options simply because WARL is not
architecturally constrained)
The other problem we have is that some behaviors are non-deterministic, but
that isn't solvable no matter how perform configuration
(e.g. the poster child of them: misaligned accesses, which can trap or be
performed, depending on non-ISA microarchitectural state that Sail can't
know about)

abukharmeh, 2022-02-03:
```
We could, of course, change that and embed it inside the Sail csim - but
what would the advantage of that be?
```
I don't have a strong preference on either options, but I am not sure how scalable its to pass all options we want to configure the model with via cmd line args, you know much better than me the extent we like to configure the model with and all possible options. At the moment the model is barley configurable, and most options we have are switched on and fixed on by default.  I just thought its neater to parse and pass data directly to the compiled Sail code. 

```
The way that the csim is currently configured is to pass command-line
arguments to Sail, and Sail compiles the csim appropriately. 
```
[image]

Maybe I misunderstood you, but at the moment as far as I know, Sail compiler compile the Sail code into C/ Ocaml code that contains both options and then the c simulator switch a flag to either enable this or that configuration, this happens in runtime as shown in this picture.  Some options like XLEN get configured at compile time, but that's the Makefile selecting the correct files for a given configuration.

allenjbaum, 2022-02-04:
The difference is if the flag is a constant that is in the csim c-code, or
a variable that is evaluated after the csim model is built and compiled.
In the former case, the csim will be run through the compiler, and the
option not selected will be eliminated completely.
In the latter case, it is evaluated after csim is compiled, so both options
are present and it has to read the value and branch to correct code.
In effect, one has a custom simulator targeted to a specific
implementation, vs. a generic one that has to continually test and select
the code that it runs.

This is pretty easy for options that have two, or perhaps 3 possibilities,
e.g. C-ext is
either implemented (all C-extension code is present and unconditional), or
is
    not implemented (all C-extension code is removed),
or is  implemented  but can be disabled (all C-extension code is present,
but can't be used if misa.c==0)

That depends on when the YAML parsing occurs; If it is called from the csim
itself, it's runtime.
If it's parsed ahead of time then the result of the parsing is passed as
command line options that set constant values, and the compiler takes it
from there.

abukharmeh, 2022-02-04:
`a variable that is evaluated after the csim model is built and compiled.` Yes, this is what I meant to say, all of the options we currently have are evaluated at runtime of the simulator, and not fixed before dead code elimination when they are inputted to the C/ OCAML compiler.
[image]

allenjbaum, 2022-02-04:
OK, I learned something new, that the options are input to csim, not an
input to Sail that builds the csim.
That means there is only one version of csim - which is golden, and not
multiple versions depending on options.
That's both good and bad.
It also makes the arguments against generating specific per/implementation
Sail code for CSR WARL field access stronger.

It doesn't really change whether the c-code that parses YAML and sets
specific variables
should be directly called by the csim, or a separate program that generates
command line options.
I still think the separate program has more flexibility and observability.

That's a minor issue compared to the complexity of WARL fields, though.

allenjbaum, 2022-02-21:
I'm replicating this comment from PR#128: (because I'm an incompetent GIT user - I'm sure there's a better way)
There are currently 3 proposals on the table for handling Sail configuration via the riscv-config YAML file:
a. This pull request: this uses a Python script to produce Sail code for the read/write functions for each CSR.
a.1.: - this means it's a different Sail model for every implementation, and that we can't ship a single golden csim model inside a docker container, it has to be recompiled every time it is used.
a.2: - it isn't clear how readable the code will be.
a.3 + it uses the latest riscv-config schema

b. The rv-config branch: This uses a separate Ocaml program that parses the YAML and produces, for each CSR, produces a Sail type which can be used to paramterize the write function.
b.1 -This is superficially like the above; I don't know that a single precompiled csim could be shipped
b.2 -The parser is written in Ocaml, which may be powerful, but is not well known and we'd have maintenance problem
(to be fair, you could say that about Sail, too...)
b.3 -It uses an older version of the riscv-config schema and would need a major upgrade

c: My proposal: this would use a python or c script that parses the YAML into very easy to decode per/CSR lists that are passed into Sail as configuration values. Sail would parse each element into XLEN width bitfields and LOG2XLEN width integers (and/or lists of them). The Sail CSR write function would treat each element (or element of a list of elements) as a rule and apply it to the CSR and the write value sequentially until the list is consumed .
c.1 - the parser doesn't exist yet
c.2 - the sail code doesn't exist yet (though it would resemble the rv-config branch support.
c.3 + no new types are needed, and only a single csim is generated

Note that all 3 proposals do the bulk of their parsing in a separate bit of (non-sail) code; the primary difference is whether individual Sail functions per CSR are generated, individual Sail types per CSR are generated, or text strings per CSR are generated.

One unknown to me is whether making changes enabling a single csim will affect other uses of the formal model; that is, does the "data dependency" of using a string (or any other basic data type) to legalize CSR writes mean that theorem proving becomes impossible? Does it affect the OCaml model?

martinberger, 2022-02-21:
_> does the "data dependency" of using a string (or any other basic data type) to legalize CSR ..._

String is a bad data-type for communicating information in Sail because you get little support from the type checker. Using sum types like enumerations / unions is much better because the type checker can do e.g. exhaustiveness checks. Using sum types is perfectly compatible with theorem proving (a sum types is closely related to logical disjunction).

martinberger, 2022-02-21:
> Spike already has this option, though from a device-tree formatted file.

If Spike already does this, maybe the Sail model should do the same, for uniformity?

allenjbaum, 2022-02-21:
I had guessed that a string is a bad data type for this, which is why I pointed it out.  I'd really like to be able to pass types rather than a string; I just don't know how to do that, or what it means to do that.

 If we have a type struct which is a list that contains a list as one of its members - how to we get that into the csim from the command line? The command line argument is a string, so some level of parsing is needed to turn it into a type - but you're still starting with a string, which is where I was coming from. 
 
 It's either my ignorance or just the point of view I'm using. Both probably. But if we can do what you are saying, I'd be really pleased. 
 
My main point was that we don't literally need to have an enormous string of command line options, as opposed to pointer to a file that has an enormous string of command line options; that was probably a bit too obvious.

 Spike has a somewhat limited capacity for configuration compared to what we think is necessary. I don't have an opinion whether we should br using DT format or riscv-config format, but we probably need a conversion tool. Not that typically, the device tree is built by boot code that (among other things) will look at the config-struct, and we do want to be compatible with that (at least by having a conversion tool).

martinberger, 2022-02-21:
@allenjbaum We do not need enormous string of command line options.  We can ask the C simulator to take an additional command line argument, which might take a file name as argument.  IIRC this requires changing [1]. The file given as argument needs parsing and the information in the file needs to be converted into a form that the C, which is compiled from Sail, can handle.

I think parsing, conversion etc	is not a difficult problem. The most interesting problem is how best to represent the configurations in Sail, ideally in a typed manner. @PeterSewell 	made some suggestions previously.

[1] https://github.com/rems-project/sail/blob/sail2/lib/rts.c#L604

allenjbaum, 2022-02-21:
That's exactly what I had in mind; I guess I'm not explaining myself well.
The legalization functions are easy; delimiting the size and position of
the implementation defined subfields are what I think of as the difficult
part.

Mapping that to the type that we need (if the type system is as flexible as
we need, as opposed to multiple types) is the part I'm unclear about.
I proposed mapping all of the struct elements to fixed sized integers or
lists of bitstrings because otherwise we'd have hundreds or even thousands
of types, and we'd have to dynamically choose them. I wasn't under the
impression that we could define new types at run time, as opposed to
filling the value of an existing type at run time.
Specifically something like:
 { [boolean COND], int6 FLD_HI,[int6 FLD_LO], {bitfield64 VAL_HI,[bitfield64
VAL_LO] }, enum MAPFUNC}. where (I'm making this syntax up)
{ } surrounds a list of arbitrary size,
[ ] surrounds an optional subfield,
int6 is a 6bit integer,
bitfield64 is a 64bit bitfield. (use int5 and bitfield32 for RV32)
will work for every CSR I can think of offhand (one of the MAPFUNC enums is
MSKVAL where VAL_HI is the mask and VAL_LO is the background value)
We can generate 4 or 8 separate types if optional subfields aren't allowed
(*2 for RV32 vs RV64)

martinberger, 2022-02-21:
Luckily Sail can define option types [1] and they are widely used in the RISCV model [2]. It looks like the type you are sketching is a list of 4-tuples,  like so:

    list( ( option(bool), int6, option(int6), T ) )

where the last component is a list of 2-tuples, like so:

    type T = list( ( bitfield64, option( bitfield64 ) ) )

Did I get this roughtly right? The precise handling of the bitfields may need some care / dependent types.

[1] https://github.com/rems-project/sail/blob/sail2/lib/option.sail#L76-L79

[2] https://github.com/riscv/sail-riscv/blob/master/model/prelude.sail#L72

martinberger, 2022-02-21:
> define new types at run time

We cannot define Sail types at (Sail or C) run-time.

allenjbaum, 2022-02-21:
That is what I thought, and makes perfect sense..
What you describe above is what I was thinking of, except I think we can't
deal with the condition field as part of this type.
The condition bool is a little tricky; in theory, it's a function of any
implementation defined architectural state,
but a) I don't think it's possible to express that in Sail, and b) in
reality it is a function of architecturally defined architectural state.
So, the CSR_WRITE function must be passed the name of the CSR, a predefined
list of booleans, and the typed variable:
  list( ( int6, option(int6), T , enum )

I need to think about how we apply the booleans to the list elements. It's
not a very common case, and it's possible it would be a 1:1 map,
but, because of WARL, we have to pre-define the number of booleans, but
can't predefine the number of list elements of the type
(an implementation might split an WARL field across multiple list elements
(e.g. a CSR containing addresses could be split into
 - a sign extension part at the high end,
 - the address part, and
 - the alignment part (e.g. 12 bits of 0 if page aligned)
Or split into more or fewer parts.

There would be two different types: , one for RV64, and one for RV32.
So the remaining part is exactly the format of the commandline argument
that the csim will be presented with, which needs a CSR nsme, and the
assignment.
There would be separate list elements for each subfield for a CSR
(delimited by the 2 or 1 int6s),
 and then the legal values for that subfield (a list of ranges or explicit
values, or treating it like mask/value pair, depending on the enum

The code for this should be pretty simple, just applying a standard
function to each named CSR for each list element, and for each T-subelement
for that element.
The pre-parser would sort and validate the list elements (ensuring that
field definitions can't overlap, for example) to produce these types from
the YAML.

PeterSewell, 2022-02-22:
>
> > define new types at run time
>
> We cannot define Sail types at (Sail or C) run-time.

And nor, as far as I can see, do we have any need to.  Let's look back
again at Robert Norton-Wright's earlier version (I'm sure that would need
to be extended or adapted, but one should start from that and a clear
understanding of what more is needed, rather than trying to start from
scratch).  First, he defines some fixed types for the configuration
parameters of that version of the model, in:

https://github.com/riscv/sail-riscv/blob/rv_config/config/riscv_config2sail/sail/riscv_config_types.sail

e.g.
struct WARL_bitmask = {
mask : nat, /* XXX consider using bits('n) */
default_val : nat
}

struct isa_config = {
I : bool, /* The base integer ISA (always true because E is not implemented
yet) */
M : bool, /* mul / div*/
F : bool, /* floating point */
A : bool, /* atomic memory */
D : bool, /* double floating point */
C : bool, /* compressed instructions */
S : bool, /* supervisor mode */
U : bool, /* user mode */
Zicsr : bool, /* control and status registers */
Zifencei : bool /* instruction fetch fence */
}

struct riscv_isa_config = {
ISA : isa_config,
hw_data_misaligned_support : bool,
xlen : nat, /* XXX 32, 64, 128 */
misa : misa_config,
mvendorid : id_reg_config,

Second, he has a standalone OCaml program to read YAML and produce a Sail
file defining several top-level configuration values such as isa_config, of
those types.

Third, the model uses those values in various places, e.g.

https://github.com/riscv/sail-riscv/blob/rv_config/model/riscv_sys_regs.sail#L176
function legalize_misa(m : Misa, v : xlenbits) -> Misa = {
if sys_enable_writable_misa ()
then {
l = [v with 25..0 = legalize_warl_bitmask(v[25..0],
isa_config.misa.Extensions)];

using a fixed legalize_warl_bitmask function that takes the relevant config
value to tell it how to legalize this field. These legalize functions are
in:

https://github.com/riscv/sail-riscv/blob/rv_config/model/riscv_sys_regs.sail#L88
e.g.
/*
THIS(current_val, written_val, legal_range) legalizes a WARL field
according to
given legal_range of values and legalising mode.
*/
val legalize_warl_range : forall 'n, 'n > 0 . (bits('n), bits('n),
WARL_range) -> bits('n) effect {escape}
function legalize_warl_range(current_val, written_val, legal_range) = {
if (is_value_in_range(unsigned(written_val), legal_range.rangelist))
then written_val
else match(legal_range.mode) {
WARL_Unchanged => current_val,
WARL_Largest => to_bits('n, WARL_range_max(legal_range.rangelist)),
WARL_Smallest => to_bits('n, WARL_range_min(legal_range.rangelist)),

So what would have to change?  AFAICS (though I say this without being
familiar with what's in the later versions of the YAML):

- one would prefer the functions of the model to be explicitly rather than
implicitly parameterised on the config value, passing down a single
parameter (of a top-level struct type containing all the config) rather
than referring to global identifiers.  That gives us a single executable
and also a single mathematical model for any proof
- one would like the config value to be constructed at runtime, by the C or
OCaml interpreter, by parsing some YAML (or whatever)
- the Sail type and the structure of the YAML file should (and should be
required to) exactly correspond for each version of the model
- the parsing can be pretty much generic in that structure
- the details of the allowed shapes of config values probably need to be
generalised in some way (I don't know what), adding more cases to the Sail
types describing them (e.g. WARL_bitmask above)
- in the cases where legalisation depends on other system state, you need
to decide what state, and pass that into the uses of the relevant
legalisation functions

Peter

***@***.***>

allenjbaum, 2022-02-22:
What is in the newer version is that the legalization function is not
fixed, and the format of WARL fields are not fixed; they can be broken up
into an arbitrary number of subfields which each have separate legalization
requirements - and those legalization requirements can also be an arbitrary
number of ranges (or individual values).

- the details of the allowed shapes of config values probably need to be
generalised in some way

So (and this is just me speaking, and I am very, very far from being
competent in Sail) the legalization function (that I've proposed) takes a
descriptor type
(a list of 4-tuple rules, one of which is a list of 2-tuples) and applies
it to each WARL field (at least, each that is larger than a bit or is
conditional).
Each list element is the MSB:LSB position of the field - or subfield, a
list of  value ranges , and an enum that chooses the legalization function

A CSR like MTVEC is architecturally  2 separate WARL fields, but it would
be completely reasonable to expect that the larger WARL field is built up
of 1,2, or 3 subfields, each with their own legality, and each of those can
have implementation defined boundaries.
So           MTVEC_MODE(mtvec.mode, write_val, legal_range)
becomes MTVEC_MODE(mtvec.mode, write_val.mode, THIS_WARL_DESCRIPTOR)
where the descriptor is one of the 4 tuples corresponding to that field.
The legalization function iterates through the ranges until it either
- it finds a match and writes the write_val or
- or doesn't, and applies the legalization function legalize(enum
curr_value, write_val, smallest, largest, closest up, closest down....)

My proposal doesn't even bother splitting the CSR into separate fields
If you have the subfields MSB:LSB, you may as well just apply all them to
the entire register, and each will only affect the corresponding subfield
defined by its MSB:LSB.
Non-WARL fields are defined by an overall mask and value (where mask
applies to RW bits, and value applies to RO and WARL fields), and
the WARL_DESCRIPTOR is the list of 4-tuples, each of which is applied
sequentially as above.

I'm not sure I  understand  "- one would prefer the functions of the model
to be explicitly rather than implicitly parameterised on the config value,
but I think that's what we want to do also

- one would like the config value to be constructed at runtime, by the C or
OCaml interpreter, by parsing some YAML (or whatever)
I understand there is a standalone OCAML parser. But there is far less
OCAML expertise than there is in Sail, much less Python or C++, or even PERL
So I not going to recommend that we augment the OCAML code and instead
recommend it be rewritten in one of the other languages above so it can be
maintained.
Since it is standalone, that shouldn't affect anything.

I am not sure what you're thinking of when you say
    - "the Sail type and the structure of the YAML file should  be required
to exactly correspond for each version of the model."
What are you thinking we would change?
Any time we change by adding a new CSR, it will be described using the same
CSR descriptor type.

- the parsing can be pretty much generic in that structure
I think that matches what we are thinking
- the details of the allowed shapes of config values probably need to be
generalised in some way
As above
- in the cases where legalisation depends on other system state, you need
to decide what state, and pass that into the uses of the
relevant legalisation functions

For the conditional cases: we are only going to support architecturally
defined conditions, not implementation defined conditions,
so we aren't deciding on the state, the spec will. It is permissable for an
implementation to decide use arbitrary architectural state, but I can't
expect Sail to support that.

 I think we have to re-define the 4-tuple in the config struct to be a
5-tuple with a condition as the [optional] added parameter (If the
parameter doesn't exist, it defaults to true.)
A CSR that has two formats depending on the value of some specific state
(or combination of state) would have two sets of rules, one with the
condition and the other with its inverse
 (or, if there are cases with more than two, then 3 rules, and only one of
them has a condition evaluating to true)
I'm just not sure now to do the mapping of condition to a specific CSR in
the code in a clean way, but it would be fixed, at least.