riscv/sail-riscv #150: Functional equivalency checking between models

billmcspadden-riscv, 2022-02-07:
I would like to start a discussion about the need for a method to do a functional equivalency check
between 2 models.

As I've started playing around with the RISC-V Sail model,  I can see places where I might want to make
(what I consider to be) benign changes in the code.   As I started doing this,  I began to realize the risk
in doing so;   I might be making something more than an innocuous change in the model,  IN THE GOLDEN
MODEL THAT DICTATES WHAT THE RISC_V ARCHITECTURE IS.    Yes,  fear and dread fell upon me

It seems to me that we really need a method that can identify functional changes between 2 models.

In the RTL world,  we have tools that allow us to verify that 2 different models (an RTL specification and a 
logic netlist) are equivalent.   This is the kind of check that I'm talking about.

I know this is not an easy problem,  but I'm pretty well convinced that we need something of this sort.

Comments are welcome,  as always.

Bill Mc.

jrtc27, 2022-02-07:
That's called the theorem prover backends? Part of the whole point of Sail is being able to generate output to feed into theorem provers and formally (in the mathematical sense, not the directed random testing sense) verify whatever you like.

billmcspadden-riscv, 2022-02-07:
RIght.   Does such a prover exist?    I certainly have no desire to  do directed random testing;  I've
left that world.

If such a prover does exist (specifically for the RISCV model),  where is it and how do I use it?

If such a prover does not exist,  how do I go about learning how to make one?

Inquiring minds want to know....

allenjbaum, 2022-02-07:
Ooh, good questions. Specifically, here, it's proving equivalency between
two *Sail* models, which might make life a bit easier.

martinberger, 2022-02-07:
Sail currently compiles to 
HOL4 https://github.com/HOL-Theorem-Prover/HOL
Isabelle/HOL  https://isabelle.in.tum.de/
Coq https://coq.inria.fr/

They are interactive provers. I imagine you want to prove equivalence between individual functions that you changed, rather than the full model (which is big).

PeterSewell, 2022-02-08:
There is also the Isla symbolic execution engine
https://github.com/rems-project/isla, which in some cases might be used for
this.

peter

martinberger, 2022-02-08:
@PeterSewell Does Isla handle unrestricted recursion?

PeterSewell, 2022-02-08:
no - but why would you want to?

PeterSewell, 2022-02-08:
> Bill wants to prove equivalence of two full models ..

Some refactorings would just be of a small part - and in any case, in most
architecture specs, there's no general unbounded recursion except (if one
has this, which one doesn't always) in the top-level loop.  Recursion down
the levels of a page table is the most I've seen, bounded by depth 5 or so.

Peter

martinberger, 2022-02-08:
I think there is a lot of recursion 'under the hood' in the infinite precision arithmetic and other helpers, that handled by C, Coq etc, depending on backend.

bacam, 2022-02-08:
We already use Isla for equivalence checking a particular optimisation inside Isla, although I think it's still limited to pure functions.  It would probably be possible to build a more general Isla-based tool for checking some notion of trace equivalence.

Recursion "under the hood" isn't a problem because Isla is passing these built-in operations on to the SMT solver.  Of course, if the equivalence is subtle enough then the solver might run into difficulty.