riscv/sail-riscv #152: Build fails on M1 Mac with C99 error.

john-terrell, 2022-02-13:
Building on M1 macOS results in errors compiling:

$ bash build_simulators.sh

...

generated_definitions/c/riscv_model_RV32.c:272182:22: error: implicit declaration of function 'convert_fbits_of_fbits' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        zgsz311930 = CONVERT_OF(fbits, fbits)(zuz32671, UINT64_C(64) , true);
[build.log](https://github.com/riscv/sail-riscv/files/8056918/build.log)

jrtc27, 2022-02-13:
Not a bug in sail-riscv but a bug in the Sail compiler itself. Works fine for me with my Sail compiler on macOS 12.1 (M1 Max) though, so maybe you just have a bad version?

closed 2022-02-13 by jrtc27

john-terrell, 2022-02-14:
Thanks for the info.   I'm using the latest sail (v0.14) on an M1 Max.   What version of sail are you using - perhaps there's a problem with newer sail?

jrtc27, 2022-02-14:
0.14. I don't think I've pinned a newer version built from Git on this machine (and `opam pin list` is empty) but it's possible.

john-terrell, 2022-02-14:
Hmmm.  I just installed sail using spam:
```shell
$ opam install sail
```

john-terrell, 2022-02-14:
Perhaps a C compiler difference if it's complaining about C99?

```shell
$ cc --version

Apple clang version 13.0.0 (clang-1300.0.29.30)
Target: arm64-apple-darwin21.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

jrtc27, 2022-02-14:
Nope:
```
Jessicas-MacBook-Pro:sail-riscv Jess% clang --version
Apple clang version 13.0.0 (clang-1300.0.29.30)
Target: arm64-apple-darwin21.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
Been a default error for a while now anyway.

john-terrell, 2022-02-14:
Huh.   That's odd then.   Ok, thanks for checking.   I'll dig on my end and see if I can figure out why sail is doing what it's doing.

john-terrell, 2022-02-14:
Another data point: I tried building using GCC instead of Clang and it also fails similarly.

jrtc27, 2022-02-14:
GCC isn't GCC, it's just an alias for Clang to work around build systems that make too many assumptions about what Unix systems look like. `gcc -v` still prints out the same Apple clang version.

Alasdair, 2022-02-14:
The error seems to indicate that Sail is generating an ill-typed assignment, assigning a 64-bit bitvector to a non 64-bit bitvector. What is your Z3 version?

Alasdair, 2022-02-14:
Also would it be possible to upload the generated C file you have?

john-terrell, 2022-02-14:
Hi, my z3 is:
```shell
$ brew info z3
z3: stable 4.8.14 (bottled), HEAD
```