riscv/sail-riscv #155: CSIM Build System

abukharmeh, 2022-03-05:
Hi, 

At the moment, the generated C definition of the simulator gets written to a single C file before being compiled. 
I wonder how realistic is it to compile the model into many C files (trying to preserve the boundaries of the modules), and then compile these files.

I think this would help with debugging, and could perhaps help speed up recompilation of the model?
@jrtc27  Any ideas if this is possible as things stands? 

Ibrahim.

scottj97, 2022-03-07:
It could speed up even a clean compile (not just incremental) if you use `make -j` to compile many C files in parallel.

martinberger, 2022-03-07:
@abukharmeh This is more a Sail issue than RISCV model. I don't think the current Sail compiler gives you separate compilation. Sail doesn't really have a module system.

abukharmeh, 2022-03-07:
I think perhaps somebody should try to come up with clear boundaries for each SAIL file in the model, @Alasdair  any thoughts from the general Sail perspective?

martinberger, 2022-03-08:
@abukharmeh What do you mean by _"clear boundaries"_ here?

Timmmm, 2024-05-07:
As @martinberger said this would require changes to the Sail compiler, so I'll close this here - feel free to reopen on [the Sail compiler repo](https://github.com/rems-project/sail) but I wouldn't hold your breath!

closed 2024-05-07 by Timmmm