riscv/sail-riscv #156: Unexpected misaligned SW

aignacio, 2022-03-11:
Hi everyone,

I was running the RISC-V compliance tests using the RISCOF [framework](https://riscof.readthedocs.io/en/stable/installation.html) and I had a single test failing which was indeed passing using the [old framework](https://github.com/riscv-non-isa/riscv-arch-test) (the one that doesn't use SAIL C simulator as reference).
[image]
[image]

For my surprise when I checked the test `disasm` and the `sail-c-simulator.log` I noticed that store (sw) of the DUT signature was coherent with the address and the one executed by SAIL-C simulator was not. For instance if you check just the first mismatch in the picture down below, we have the following scenario:

```bash
[101] [M]: 0x80000138 (0x7EB52FA3) sw a1, 2047(a0)
mem[0x80002216] <- 0x00080000
```

As the final address is equal to 0x2 `(0x8000_2216 - 0x8000_2214[w-aligned] = 0x2)`, I'd expect the simulator to write 0x00 in the upper 16-bits of the address and not in the 0x8000_2214 as you can read in the attached simulator signature.

[image]

Is it expected such behaviour from the SAIL-C RISC-V simulator? If so, how it's computed the final LSU address and the sw?

aignacio, 2022-03-21:
discovered it's a design issue

closed 2022-03-21 by aignacio

allenjbaum, 2022-10-11:
This is what I thing is going on:
ra is loaded with 80002210 according to the disassembly.

Inst_0 tries to store  the low 4 bytes of 40000 at address 80002211,

so that overwrites 213,212,211 in the first line with 04,00,00, leaving 210
as EF ,

and overwrites 214 in the second line  with 00, leaving 217,216,215 as DE
AD BE

Inst_1 tries to store  the low 4 bytes of 80000 at address 80002216,

That should overwrite 217, 216 in the second line with 00,00, and 219,218
in the 3rd line with 00,08

which is precisely what I see Sail doing.

What I think your DUT is doing is storing the LSB part of the unaligned
write, but skipping the MSB part that crosses the 4byte boundary.

This a nasty, but legal architectural artifact of RISC-V unaligned access
definition.

It permits 3 different outcomes of an aligned store: store all, store the
first part and trap, or store nothing and trap.

SAIL can be configured to handle all or none, but doesn't handle the middle
case yet.

It's actually worse than that, because the RISC-V architecture allows all 3
behaviors for the same address.

That is, sometimes an implementation will trap, and sometimes it won't --
depending on non-architectural state

(e.g. the TLB entry for the MSB part is valid, or the cacheline of the MSB
part is valid)

That is non-deterministic from an architectural point of view, and SAIL
cannot model that by definition.

Our plan is to eventually try to deal with that case by running the Sail
model 3 times,

and declaring a match if any of the 3 behaviors is seen in the result.

That turns out to be much more difficult than it sounds, unfortunately,

and we will need a lot of test changes to accommodate non-deterministic -
yet legal-  behavior.