riscv/sail-riscv #159: Float coversion returns incorrect value, for the signed (+/-) raw integer inputs

ptprasanna, 2022-04-08:
Issue Description: With reference to the attached log, at PC 260 we are expecting the value 0xFFFFFFFFDE924770 to be stored, but rather the sail is storing 0xFFFFFFFF5E924770. like the same at PC 274 and 284 the actual value is different than the expected. Also we did observe that the issue might be around the sign bit. Have attached the log and disassembler, please let me know if more information would help.

Log - https://github.com/ptprasanna/IssuesAndBugs/blob/main/fcvt.s.l_b25-01.log
Disass - https://github.com/ptprasanna/IssuesAndBugs/blob/main/ref.disass

ptprasanna, 2022-05-10:
This issue is more of an issue with the generator, and also it is being tracked under https://github.com/riscv-software-src/riscv-isac/issues/38 and https://github.com/riscv-software-src/riscv-isac/issues/39
Hence Closing this issue.

closed 2022-05-10 by ptprasanna

allenjbaum, 2022-10-11:
Your log is not the signature. The signature - which is what gets stored to
memory, doesn't have all those 0xFFFFFFFF in the upper half.
That is an artifact of the Nan boxing that has no effect since an RV64F
can't store those upper bytes.

The test data is a bit odd; it's loading an 8B integer, but the test data
is repeated 4 byte data, but offset by 4 bytes.
so (00, -XY),( -XY,  -XY), ( -XY,  -XY),( =XY, +XY), ( +XY, +XY),  ( +XY, +
ZZ)
The signature contains 3 negative values, followed by 3 positive
values ( DE924_770,
&  5E924_770) for inst_1..inst4, and different values for inst_0 and inst_5.
That's exactly what I would expect, so it doesn't sound like a Sail problem

Sail and Spike give exactly the same result, so I think your expected value
is wrong, or your timing is off-by-1.
so you're looking at the test before or the test after what you tink it is.