riscv/sail-riscv #161: Code bloating in the generated coq file

xxuejie, 2022-05-11:
Hi, I was trying to leverage the generated Coq file for theorem proving. I tried generating the coq file using:

```
make riscv_coq
```

(Tho I want to mention that current support for Coq has a few quirks, so I end up using this [branch](https://github.com/xxuejie/sail-riscv/tree/coq-fixes) with some fixes).

However one thing I noticed, is that the generated file is far too big, `riscv.v` is 15MB, or 115258 lines of code. The compiling phase for this file, is also super long: it's been 22 hours since I started the compiling phase on a Ryzen 3900x machine yet it is still running.

I do understand that much of this could be a general sail issue(generated file for [sail-arm](https://github.com/rems-project/sail-arm/blob/master/arm-v8.5-a/snapshots/coq/aarch64.v) is also quite big), but I do want to first check here: am I doing things the correct way? Or is there a way to truncate certain specs that are not needed, so the generated file can be much smaller and compiled faster?

xxuejie, 2022-05-12:
Some update: after about 46 hours, the following error was reached:

```
File "./generated_definitions/coq/RV64/riscv.v", line 43576, characters 0-43504:
Error: Stack overflow.

make: *** [Makefile:398: generated_definitions/coq/RV64/riscv.vo] Error 1
```

I'm doing this on a machine with Ryzen 3900x + 64G memory.

My question here is: is there a way now to choose to only build certain parts of the spec? Or is there any plan to make the spec modular in the future?

bacam, 2022-05-12:
You've got the right idea, and the changes you've made in your branch are fine - I've done the same but hadn't committed them yet because I hadn't checked them with the current release of Sail.  The main problem is that we tried recreating the type checks from the Sail model in the generated Coq that check (for example) that bitvector slicing operations are in range, but the tactics to recreate these proofs are rather more costly and fragile than we hoped.  The size of the generated Coq file is misleading - a lot of that is unnecessary whitespace due to limitations of the pretty printer and Coq happily ignores it.

The easiest way to get a working copy at the moment is to look at the virtual machine version of the artifact for our upcoming PLDI paper, at https://doi.org/10.5281/zenodo.6320641, which includes a copy of the model in `Desktop/islaris_artifact/_opam/lib/coq/user-contrib/RV64/`.  (You might also be interested in the approach taken in the paper, which uses traces from symbolic execution of the Sail model rather than a direct translation into Coq.  The Coq model is included to demonstrate that the traces can be shown to match the model.)

We intend to make these models more stable and usable based on our experience from that work, although before that I need to update some parts of the Coq backend to track other recent changes in Sail.  That said, if you have a particularly urgent need let me know.  Two things that can help are to comment out the encoder/decoder definitions if you don't need them, and to use coqc's `-time` option to identify where compilation is getting stuck.

xxuejie, 2022-05-13:
Thanks a lot! This makes perfect sense, islaris does look like a good solution in this sense, I will definitely give it a try here.

closed 2022-05-13 by xxuejie