riscv/sail-riscv #162: For RV64, the odd numbered PMP configuration registers are NOT illegal

UmerShahidengr, 2022-06-02:
Hello,
For RV64, the odd numbered PMP configuration register access should be illegal. I have written a test to verify it on Sail-riscv but it is giving access to pmpcfg1, pmpcfg3 etc. The test for riscof environment is attached.

[PMP-OddCSR-access.txt](https://github.com/riscv/sail-riscv/files/8821730/PMP-OddCSR-access.txt)
.

scottj97, 2022-06-02:
[This code](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_insts_zicsr.sail#L108-L111z) is where the odd-numbered registers are excluded.

The code looks correct to me, and in my environment, I can see `pmpcfg1` taking an exception as expected.

allenjbaum, 2022-06-03:
My guess is that Sail thinks it is running RV32 instead of RV64.
I assume that same code exists for the 64 PMP entry option> Or perhaps that
Sail support hasn't been added yet.

UmerShahidengr, 2022-06-04:
At my side, Sail is not giving any exception or write but it is returning all 0s after read.

Abdulwadoodd, 2022-06-24:
> [This code](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_insts_zicsr.sail#L108-L111z) is where the odd-numbered registers are excluded.
> 
> The code looks correct to me, and in my environment, I can see `pmpcfg1` taking an exception as expected.

Odd number-numbered PMP configuration CSRs are excluded [here](https://github.com/riscv/sail-riscv/blob/master/model/riscv_pmp_regs.sail#L150-L153z) as well while modeling the PMP registers.

allenjbaum, 2022-06-24:
The only explanation I can think of is that your model is really an RV32. Try running a test of addw and see if it traps (and ensure that the test is actually executing and not filtered by riscof). If it traps, then you're running an RV32 config

Abdulwadoodd, 2022-06-28:
My model is RV64, I have rechecked it. Apparently, the sail is right here as all the PMP CSRs are WARL. Though, access to odd CSRs is illegal but it does not say anything about raising an exception on odd CSR. Spec also says `All PMP CSR fields are WARL and may be read-only zero`. which is exactly what sail is doing.

scottj97, 2022-06-28:
It is supposed to take an illegal instruction exception.

During execution of a CSRR instruction, the [execute clause](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_insts_zicsr.sail#L259) calls [check_CSR()](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_sys_control.sail#L211) which calls [is_CSR_defined()](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_sys_control.sail#L77) which [returns false](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_sys_control.sail#L101-L104) for `pmpcfg1` under RV64, which results in an [illegal instruction exception](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_insts_zicsr.sail#L260).

I would suggest adding some printfs in these points in your model to see where things are going awry.

allenjbaum, 2022-06-28:
I have confirmed that this WARL interpretation is incorrect and scottj97's interpetation is correct.
Odd numbered PMPCFG CSRs don't exist/are not implemented, and accessing unimplemented CSR  must trap with in illegal instruction cause. So, yes, all PMP CSRs are WARL, but that doesn't apply to non-existent CSRs, just as any access to PMPADDR16-63 must trap if only 16 PMP regions are implemented

allenjbaum, 2022-07-29:
Do move this along: have we bottomed out on whether there is a bug in the model, or whether Abdulwadoodd is somehow misconfiguring something? I'm getting mixed messages here

Abdulwadoodd, 2022-07-29:
Your explanation above is correct and I was misinterpreting the spec. Only 16 CSRs are implemented and accessing unimplemented CSRs raise exception as Scott pointed out how Sail does that.

allenjbaum, 2022-07-30:
I thought you weren't trapping, and you thought it should.  You said "Sail is not giving any exception or write but it is returning all 0s after read." If that is still the case, -- do you know why your version is doing that? Please close the issue if you are able to get the correct results.

closed 2022-08-03 by UmerShahidengr