riscv/sail-riscv #163: Changing PMP granulairy in Sail

UmerShahidengr, 2022-06-07:
What is the standard method to change granulairy of PMP in Sail model?
I tried to change granulaity by writing 8 instead of 4 in [this line](https://github.com/riscv/sail-riscv/blob/master/c_emulator/riscv_sim.c#:~:text=int%20signature_granularity%20%3D%204%3B) But it does'nt do anything.

bilalsakhawat, 2022-06-07:
sail-riscv does not support granularity > 4. [link](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/model/riscv_pmp_control.sail#L71)
And the line you changed is for signature granularity.

UmerShahidengr, 2022-06-07:
When will that support be added in sail-riscv?

bilalsakhawat, 2022-06-07:
The rv_ram_base is 0x80000000. If you try to set up a region below this address you will get exceptions. And rv_ram_size is 0x800000. So, the upper bound is rv_ram_base + rv_ram_size, [link](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/c_emulator/riscv_sim.c#L543). 

For better understanding look into issue #73 and pull request #151

UmerShahidengr, 2022-06-07:
Dear All,
I have written a basic test of defining a PMP region with no-write access and tried to write data in it. sail-riscv did not give expected output as it allowed the write access to reserved region (spike worked as expected and riscof environment returned fail status). Does PMP need major revision in sail model? Riscof based test (for RV32), sail-signature, and spike-signature is attached for reference.
[PMP-protection-test.txt](https://github.com/riscv/sail-riscv/files/8852896/PMP-protection.txt)
[DUT-Spike-signature.txt](https://github.com/riscv/sail-riscv/files/8852905/DUT-Spike-signature.txt)
[Reference-sail_c_simulator-signature.txt](https://github.com/riscv/sail-riscv/files/8852910/Reference-sail_c_simulator-signature.txt)

bilalsakhawat, 2022-06-07:
Do you enable pmp while running test on sail ? There is a CLI for enabling the pmp? [link](https://github.com/riscv/sail-riscv/blob/50f4bbaa5bd397ff5a71b7b39fa543d656934b80/c_emulator/riscv_sim.c#L112)

UmerShahidengr, 2022-06-08:
@bilalsakhawat I am running the test on riscof. How can I add additional command line argument in riscof command?

bilalsakhawat, 2022-06-08:
> @bilalsakhawat I am running the test on riscof. How can I add an additional command-line argument in riscof command?

RISCOF uses [riscof-plugins](https://gitlab.com/incoresemi/riscof-plugins) to run a test on the sail-riscv.  You can add command-line argument [here](https://gitlab.com/incoresemi/riscof-plugins/-/blob/master/sail_cSim/riscof_sail_cSim.py#L148).

UmerShahidengr, 2022-06-08:
Thanks @bilalsakhawat , I have updated [the following line](https://gitlab.com/incoresemi/riscof-plugins/-/blob/master/sail_cSim/riscof_sail_cSim.py#L148) to:
execute += cmd + ' --enable-pmp --test-signature={0} {1} > {2}.log 2>&1;'.format(sig_file, elf, test_name)

Now the tests are passing and PMP of sail-riscv is working fine.

closed 2022-06-08 by UmerShahidengr