riscv/sail-riscv #171: xperm4 instruction inconsistent with pseudo-code in specification

tzwaenn, 2022-07-15:
For xperm4, the pseudo code in the spec reads
`function xperm4_lookup (idx, lut) = {`
`(lut >> (idx @ 0b00))[3..0]`
`}`
and
`result[i+3..i] = xperm4_lookup(X(rs2)[i+3..i], X(rs1));`

The Sail model has 
`    result[i+3..i] = if 8*index < sizeof(xlen)`
`                     then rs1_val[8*index+3..8*index]`
`                     else zeros()`

I would expect to see 4*index here for xperm4 (as there are only 2 zeros appended in xperm4_lookup  versus the 3 zeros in xperm8_lookup).

jrtc27, 2022-07-15:
I think Spike agrees with you too. @ben-marshall is this consistent with your understanding?

tzwaenn, 2022-07-15:
I actually came to this through an implementation of [Zkn](https://github.com/chipsalliance/rocket-chip/pull/2950). Running formal verification with the OneSpin processor verification app showed a discrepancy against the Sail model and I agreed with the Zkn implementation as well.

tzwaenn, 2022-10-18:
Any updates here? How to move on if the original author is no longer responsive?

tzwaenn, 2023-01-04:
Seems there is a pull request to fix this #186

ben-marshall, 2023-01-04:
Hello! Apologies for disappearing, my RISC-V (and consequently Github) involvement tapered out over the last year so I didn't see this originally.

I did not write the Sail implementation of `xperm4`, that was @bilalsakhawat judging by the Git history.

I agree there is a problem with the Sail and I believe Spike is correct as @jrtc27 points out. IMO, solution is to simply merge #186 .

billmcspadden-riscv, 2023-01-09:
Hi Sven.   Bill McSpadden here.

We'll get this scheduled to be fixed.   We have our regular meeting
tomorrow (Tuesday),
and Bilal usually attends.  We'll discuss it then,   It appears to be a
simple enough fix.

Bill Mc.

tzwaenn, 2023-02-06:
Looks like the fix is in - resolve?

closed 2023-05-26 by tzwaenn