riscv/sail-riscv #176: Error when building the model

haitianzhishang, 2022-08-17:
[image]
After installing the sail by follow install.md from sail repo, ran make to build risc-v model and this error came. 
I'm really not sure about the reason for this error and hope for some directions. Thanks a lot

bacam, 2022-08-17:
It looks like the Makefile expects the GNU version of sed and your system has a BSD one.  Could you try the changes in #177?  I think that should work but I don't have a BSD-style system set up for building sail-riscv at the moment.

haitianzhishang, 2022-08-17:
> It looks like the Makefile expects the GNU version of sed and your system has a BSD one. Could you try the changes in #177? I think that should work but I don't have a BSD-style system set up for building sail-riscv at the moment.

error fixed. Thanks a lot!

closed 2024-05-07 by Timmmm