riscv/sail-riscv #180: Question about isa extension

haitianzhishang, 2022-09-07:
I would like to try to add some instructions based this RISC-V model. But I don't know how to debug it, I add some print information but don't know how to run it , what I can do is just running the "make check" to correct syntax errors. 
Is there some guidelines for debugging or testing? Really hope for it and thanks a lot

martinberger, 2022-09-12:
By syntax errors, you mean errors in the *Sail* syntax? If so, the Sail compiler should flag those, no?

haitianzhishang, 2022-09-15:
>

haitianzhishang, 2022-09-15:
> By syntax errors, you mean errors in the _Sail_ syntax? If so, the Sail compiler should flag those, no?

For example, I add a new instruction in this model. How could I test this new instruction using Sail?

closed 2022-09-15 by haitianzhishang

martinberger, 2022-09-15:
Given that it's a new instruction, you'd probably have to create your own tests, or, provide a second implementation, e.g. in Spike or QEMU and run them against your updated Sail model.