riscv/sail-riscv #182: switch to disable tlbs

neelgala, 2022-09-12:
I believe there is some sort of a TLB like structure used in SAIL which prevents PTW to be triggered for each memory/instruction access. It would be beneficial if we can have a switch via the cli to disable the TLBs so that all memory/instruction accesses undergo a PTW by default.

This requirement comes from the SIG-ACT with regards to capturing ISA coverage for tests dealing with translation schemes.

Timmmm, 2024-12-30:
Can you explain more why this is needed (assuming it still is). If tests want to guarantee a page table walk can't they just do an sfence.vma? Doesn't seem like the ACTs should rely on microarchitectural details about the TLB.

allenjbaum, 2025-01-03:
Good question.  ACTs are trying to avoid microarchitectural details (e.g the presence of a TLB or it's organization and behavior), not rely on them.  I think it was to ensure that certain implicit accesses were made properly (i.e.the full table walk). Yes, we could modify the tests to add the fence op, but it would be a lot of work. 
If this has already been implemented, I'd rather it continue for now, and I'll file an issue to add the fence.vma before every test access for the VMM tests.  And

Timmmm, 2025-01-03:
I think I'd be more in favour of just fixing the tests, but I'm not sure I fully understand the issue. I'll tag this for discussion in the meeting.

closed 2026-03-15 by Timmmm