riscv/sail-riscv #183: Support for decomposed-store misaligned PMP behavior

allenjbaum, 2022-09-26:
A case where a store matches 2 different PMP entries (where some of the address matches one, and some of the access matches another) then the following behaviors are allowed and should be configurable (assume misaligned accesses are supported):
- succeed if the access is decomposed and both entries allow writes
- write nothing,      trap with cause "misaligned store" if the access is not decomposed ,         xtval pointing to effective address
- write nothing,     trap with cause "access fault" if ​neither PMP entry allows writes, ​                xtval pointing​ to upper half
- write nothing,     trap with cause "access fault" if ​neither PMP entry allows writes, ​                xtval pointing​ to lower half
- write ​upper half, trap with cause "access fault" if ​lower half PMP entry doesn't allow writes, ​xtval pointing​ to lower half
- write ​lower half, trap with cause "access fault" if ​lower half PMP entry doesn't allow writes, ​xtval pointing​ to upper half

Effectively, the configuration is that 
 if misaligned access are supported or not
 if stores that cross PMP entry boundaries are decomposed 
 if so order in which those decomposed accesses are made (lower first, upper first),
    and whether both halves are checked before and store is made.

Implementations could, in theory, be neither consistent nor deterministic in their behavior, but I think it is reasonable to just configure the most likely and ask for a waiver if there are deviations.

pawks, 2022-09-27:
I don't think it is possible for the implementation to specify which of the accesses is made first. It might depend on micro-architectural states such as cache hit, tlb hit or llc hit etc. It is highly dependent on the nature of the cache(blocking/non-blocking), the design of the pipeline, the stage at which pmp checks are performed etc.

allenjbaum, 2022-09-27:
I understand, but most implementation will try one first; it may still get reordered, but this covers the common case.
If it is nondeterministic, then they can come back and ask for a waiver. That's less work for us than trying to actually handle non-determinism

allenjbaum, 2022-09-28:
Andrew has pointed out (and this matches my reading in every respect): 
(what is stored in TVAL)  cannot unconditionally be the effective address, as that might not be the address of the portion that faulted.  (e.g. on a page-spanning access, if the fault occurs on the second page, it would be incorrect to report the effective address)

My interpretation of the spec is that the address of any byte within the faulting portion is a valid thing to place into mtval.  In practice I assume this is always going to be the lowest-numbered byte in the faulting portion, but I don't see any text enforcing this property.
-------------
The two cases that are complicated are:
 - if both halves fail, which to we report?
   -- it is technically not specified, it could be either, so we could report either. To make it easy for riscof, though, 
non-deterministic  implementations  specify the common case, either  the lowest byte in the first half, or of the second.
That might not work in all case, as I mentioned above, but will usually work, and its up to the implementor to do the work explaining why their implementation is legal, got a correct answer, and should be granted a waiver
 - if only one fails, does the other (in the case of a store) store the result or not?
  -- we can treat this the same way; the implementor specifies an ordering, as above, and whether the store operation stops at the first failure (in that order) or continues to store the 2nd half. 
The handler would not know this uarch details won't know that anything did get stored, and would have to assume it didn't,
So it would effectively have to reexecute both halves, which if for cached/coherent memory but not for IO with side effects.
Coe should know better to do misaligned store to I/O, so that's likely to be fatal, in any case.