riscv/sail-riscv #193: mcycle value does not increment in sail

1danish13, 2022-11-23:
I am writing test that checks mcycle value increments. The value of mcycle increements for Spike but the valueof mcycle never changes for Sail. Here is the code

rmn30, 2022-11-23:
Hi. this looks like a problem with the integration of your tool (riscof?) with the Sail model. In the c emulator there is a call to the Sail function tick_clock: https://github.com/riscv/sail-riscv/blob/master/c_emulator/riscv_sim.c#L935 implemented at https://github.com/riscv/sail-riscv/blob/master/model/riscv_platform.sail#L319 . You'll need a similar call in whatever is driving the Sail.

neelgala, 2022-11-23:
@rmn30 within riscof, the C emulator is simply run as a command line similar to below [(see this)](https://gitlab.incoresemi.com/core-verification/riscof-plugins/-/blob/master/sail_cSim/riscof_sail_cSim.py#L148): 
```
riscv_sim_RV32 --test-signature=ref.signature test.elf > test.log 2>& 1
```
where riscv_sim_RV32 is the c emulator binary built from this repo.

Do you suggest we build the emulator differently or use other flags in the cli for the mcycle issue ?

rmn30, 2022-11-23:
Another potential solution would be to move the call to tick_clock outside the if statement, assuming the clock should increment once per instruction.

neelgala, 2022-11-23:
Just curious - any reason why its 100 as of now ? would making the default to 1 break anything internally ?

A cli arg would definitely be best way forward.

rmn30, 2022-11-23:
> Just curious - any reason why its 100 as of now ? would making the default to 1 break anything internally ?

I'm not really sure. Looks like @pmundkur wrote it. Maybe something to do with what spike did or with booting OS more quickly?

billmcspadden-riscv, 2022-11-23:
FYI:  I intend to make this configurable at some point in the future (based on settings in the RISCV-Config file probably).  

Note: There are other counters/timers that suffer from a similar problem.  The Sail model has no concept of clock cycles or of time (which is the main source of the problem).

allenjbaum, 2022-11-23:
I don't think there is any value with making it anything but 1, actually.
DUTs will have nondeterministic values that could be both less than 1 (e.g.
superscale, so multiple ops per cycle) or >1 (stalls),
so there isn't much point in making it configurable that I can see.
For tests that care about timing, we will need to use instret as a
reference, not cycle (or time).

PR #881 (2025-04-24): Change default instructions_per_tick to 2
Fixes #193. This parameter controls how many instructions need to complete for the cycle CSR to increment. While this is implementation defined (and likely not constant even within a given implementation), a default of 2 seems a lot more natural than 100.
closed 2025-04-24 by Timmmm