riscv/sail-riscv #194: Build warnings in model/riscv_pmp_regs.sail

ptomsich, 2022-11-30:
Building 4d05aa1698a0003a4f6f99e1380c743711c32052 from source outputs warnings for incomplete matches in model/riscv_pmp_regs.sail:
```
Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:144:7-12
144 |  then match n {
    |       ^---^
    | 
Most general matched pattern is 0

Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:150:7-12
150 |  else match n { //  sizeof(xlen) == 64
    |       ^---^
    | 
Most general matched pattern is 0

Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:170:7-12
170 |  then match n {
    |       ^---^
    | 
Most general matched pattern is 0

Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:193:7-12
193 |  then match n {
    |       ^---^
    | 
Most general matched pattern is 0

Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:144:7-12
144 |  then match n {
    |       ^---^
    | 
Most general matched pattern is 0

Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:150:7-12
150 |  else match n { //  sizeof(xlen) == 64
    |       ^---^
    | 
Most general matched pattern is 0

Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:170:7-12
170 |  then match n {
    |       ^---^
    | 
Most general matched pattern is 0

Warning: Possible incomplete pattern match at [model/riscv_pmp_regs.sail]:193:7-12
193 |  then match n {
    |       ^---^
    | 
Most general matched pattern is 0
```

ptomsich, 2022-11-30:
/cc @pmundkur 

Let's cc someone with knowledge of the affected lines (my best guess from looking at a `git blame`)...

jrtc27, 2022-11-30:
This is a long-standing issue that’s been there for years before the model was adopted as official

Alasdair, 2022-11-30:
I have a new release of Sail that will hopefully be merged into opam soon: https://github.com/ocaml/opam-repository/pull/22553

It has a re-written pattern completeness checker that should be much more accurate.

closed 2023-03-06 by martinberger