riscv/sail-riscv #195: Sail latex generation

haitianzhishang, 2022-12-02:
Are there any directions for using Sail latex generation? 
How can I extract  the instruction encoding information and CSR information from Sail model?

jrtc27, 2022-12-02:
> Are there any directions for using Sail latex generation?

Run `make latex`

> How can I extract the instruction encoding information and CSR information from Sail model?

There isn't any automation for that.

allenjbaum, 2022-12-02:
Why extract it from the Sail model, as opposed to a disassembler?

haitianzhishang, 2022-12-05:
> Why extract it from the Sail model, as opposed to a disassembler?
> […](#)

After adding some self-defined instructions or CSRs, I hope to extract the instruction encoding information from sail model automatically and use it to update ISA manual.

jrtc27, 2022-12-05:
You don't get those, though I don't see why you'd want them, they're there as infrastructure not as anything you'd actually want to put in a document

jrtc27, 2022-12-05:
No bug report here (and if there is one it's in Sail tooling not the model)

closed 2022-12-05 by jrtc27