riscv/sail-riscv #201: Status messages appear on stderr

ssecatchseagate, 2023-01-11:
Several status messages appear on stderr that are not errors when simulating the compatibility tests, including which file is used for the signature, as well as messages about memory allocation:
using ... for test-signature output.
[Sail] Allocating new block ..
[Sail] Allocating new block ...
tohost located at ...

As these are not errors, please move them to the log file. If they can not live there, please consider opening up an additional file for them to write to.

jrtc27, 2023-01-11:
The log file is the instruction trace, whereas this is verbose information about the model and environment itself. If you don't want them, redirect stderr?

ssecatchseagate, 2023-01-11:
How do I tell when something went wrong with the simulator if I need to visually inspect stderr as it does contain actual error messages? Ie:
c_emulator/riscv_sim.c:    fprintf(stderr, "Cannot create terminal log '%s': %s\n", term_log, strerror(errno));
c_emulator/riscv_sim.c:    fprintf(stderr, "Cannot gettimeofday: %s\n", strerror(errno));
 
I'll note that there is a SUCCESS generated in the regular log stream, but grepping for that is also not ideal (and due to some other reporting, it's not always the last line of the log file either)

If we need expected status messages outside of the log file that aren't errors, then I think that they would be better served by appearing in their own file.

jrtc27, 2023-01-11:
Check the exit code like any other executable?

ssecatchseagate, 2023-01-11:
This seems like a useful error message that I want captured and the ability to detect and resolve due to a non-zero stderr file size
It does not result in a termination of the simulation, so will not generate an exit code.

c_emulator/riscv_platform_impl.c:
// Provides entropy for the scalar cryptography extension.
uint64_t rv_16_random_bits(void) {
  // This function can be changed to support deterministic sequences of
  // pseudo-random bytes. This is useful for testing.
  const char *name = "/dev/urandom";
  FILE *f = fopen(name, "rb");
  uint16_t val;
  if (fread(&val, 2, 1, f) != 1) {
    fprintf(stderr, "Unable to read 2 bytes from %s\n", name);
  }
  fclose(f);
  return (uint64_t)val;
}

jrtc27, 2023-01-11:
That should result in termination, it's a bug that is does not

ssecatchseagate, 2023-01-11:
Honestly, I think there is a bug in that code anyway- I believe that clause will also trigger on EOF, which is an expected file condition, and should not generate that message. 
(I mean, you could generate a "detected EOF" message for that case, but I still don't want that in stderr.. 
And then something like a static to count the total number of bytes read over all time and a report of that would be lot more useful to ensure that it matches my expected file size, rather than just a note that the end was hit. 
But again, these are second order usability messages, and not in the scope of this issue)

In any case, stderr conventionally contains warning, error, and fatal messages. If there is something there, I can figure out what is the problem. Since a warning normally results in a successful exit code, that's my only way to know that something suspect has occurred. 
(I would assume errors and fatal should generate failing exit codes, and my quick scan of riscv_sim.c suggests that many fprinft(stderrs...) for fatal conditions are indeed immediately followed by exits with non-zero return codes)

Since stderr is cluttered with status messages, it needs to be parsed or inspected for every single testcase to ensure that nothing has gone wrong. 
(I'll note that if there is a warning in stderr.. and indeed it's acceptable warning for my environment and I'm unable to resolve it from my side, then, ideally, I'd also like a way to demote those to as status message, but again, that's a whole other issue)

jrtc27, 2023-01-11:
> Honestly, I think there is a bug in that code anyway- I believe that clause will also trigger on EOF, which is an expected file condition, and should not generate that message.

EOF is not expected for /dev/(u)random, they are infinite streams of random bytes.

> In any case, stderr conventionally contains warning, error, and fatal messages. If there is something there, I can figure out what is the problem. Since a warning normally results in a successful exit code, that's my only way to know that something suspect has occurred.

This is not true, it is also for informational messages. Take a tool like FFmpeg, for example, which is _extremely_ chatty, or just Git itself which prints progress information.

> Since stderr is cluttered with status messages, it needs to be parsed or inspected for every single testcase to ensure that nothing has gone wrong.

Parsing stderr is almost always a sign that you're doing something wrong; in this case, it's that the exit code isn't being used (and isn't reliable in a couple of cases).

ssecatchseagate, 2023-01-11:
Ah, well, in that case, I concur that it is a bug, and that message should be followed by a termination as you've said.

So, it sounds like this issue should branch into a another issue which is a review for fprintf(stderr..) and ensures that all error conditions terminate with a non-zero exit code. 

I still would prefer an empty file, and status messages elsewhere, but I understand your reasoning. 
Might a compromise be to tag the those messages with a state of INFO/WARNING/ERROR/FATAL or something similar, which is common for EDA tools and verification methodologies? As part of that review, any ERROR/FATAL would need to be followed by an exit, and any INFO/WARNING would allow execution to continue, so the message itself also serves as documentation for the code.
 
That would allow a run script to filter out all the INFO messages in order to ensure that no warning messages that might need further action are located in the status file? Furthermore, when the simulation has errored out, the more verbose ERROR/FATAL messages can be collected and triaged in a more efficient manner in order to determine next steps.

jrtc27, 2023-01-11:
"warning/error: " prefixes are generally how you do it in the open-source software world (and nothing for info). Distinguishing between error and fatal doesn't make sense unless you have errors that can be continued past but will still cause the exit code to be non-zero, which I don't think you want in this case. So yes, having some uniformity across all printed messages would be a good thing.

ssecatchseagate, 2023-01-11:
Distinguishing between Error and Fatal likely makes more sense from a verification perspective. 
If there is something that should happen with the DUT and doesn't that's an Error. I might want the simulation to continue to see if the problem eventually resolves, or maybe it's a known problem that I haven't disabled and want to see if others show up as part of that failure, or if the same failure happens multiple times within a simulation that might be useful information. For most verification methodologies, there is a setting that configures how many Errors are allowed before termination. In either case, it's a failed test.
If something happens with the simulator/testbench itself and the execution environment can no longer be trusted, then it's a Fatal, and execution must halt immediately. 

There appeared to be several checks on the Spike model, which seemed more likely to be Errors rather than Fatals using that use case model, though the simulation does terminate. If Errors should always terminate, then as you've said, differentiating doesn't really make sense. 
    if (!matched) {
      fprintf(stderr, "Provided DTB does not match Spike's!\n");
      exit(1);
    }

What are the next steps here? Should I file a new issue on message uniformity? Or can this serve as that issue? 
Do we need a new issue for items that appear that they should terminate and they don't?

jrtc27, 2023-01-11:
I don't see value in allowing errors that don't immediately exit, those make sense for things like compilers, not simulators.

Alasdair, 2023-01-11:
I think from my side at least the Sail memory allocation messages are unnecessary chattiness. I can reduce that and see about making the formatting of errors from the Sail runtime itself more consistent.

rmn30, 2023-01-11:
The messages about allocating memory actually come from Sail's runtime library: https://github.com/rems-project/sail/blob/sail2/lib/rts.c#L174 . Arguably these shouldn't exist or there should at least be a way to disable them or redirect them.