riscv/sail-riscv #203: packw does not properly sign extend

ssecatchseagate, 2023-01-12:
riscv_insts_zbkb.sail:
function clause execute (ZBKB_PACKW(rs2, rs1, rd)) = {
  assert(sizeof(xlen) == 64);
  let rs1_val = X(rs1);
  let rs2_val = X(rs2);
  let result : bits(32) = rs2_val[15..0] @ rs1_val[15..0];
  X(rd) = EXTZ(result);
  RETIRE_SUCCESS
}

riscv.v:
Definition EXTS {n : Z} (m : Z) (v : mword n) `{ArithFact (m >=? n)} : mword m := sign_extend v m.
Definition EXTZ {n : Z} (m : Z) (v : mword n) `{ArithFact (m >=? n)} : mword m := zero_extend v m

From the spec:
3.19. packw
...
Description
This instruction packs the low 16 bits of rs1 and rs2 into the 32 least-significant bits of rd, sign extending the
32-bit result to the rest of rd. This instruction only exists on RV64 based systems.
Operation
let lo_half : bits(16) = X(rs1)[15..0];
let hi_half : bits(16) = X(rs2)[15..0];
X(rd) = EXTS(hi_half @ lo_half);

jrtc27, 2023-01-12:
#185. Like the issue you just filed, _please check for duplicates_ before you report bugs.

closed 2023-01-12 by jrtc27

ssecatchseagate, 2023-01-12:
So, given my last mistake, I did check both the Open and Closed list for this Issue (as I thought I had done for the last one, too)
I see they are both now under Pull Requests. My apologies.