riscv/sail-riscv #204: How were the tests compiled?

Timmmm, 2023-01-19:
I haven't actually tried this but if you look at the tests, e.g. [this one](https://github.com/riscv-software-src/riscv-tests/blob/79064081503b53fdb44094e32ff54a3ab20a9bf2/isa/rv64ui/and.S) it uses `TEST_PASSFAIL` which [calls](https://github.com/riscv-software-src/riscv-tests/blob/master/isa/macros/scalar/test_macros.h#L739) `RVTEST_PASS` (if it passed). 

However if you look at the `riscv-tests` code they never define `RVTEST_PASS`. I think you have to do it yourself. If you look around `grep.app` there are a few examples, e.g. [here](https://github.com/syntacore/scr1/blob/master/sim/tests/common/riscv_macros.h#L179).

As far as I can tell the one used in this repo writes to a memory mapped IO device called HTIF (Host-Target InterFace?) at 0x80001000. The code to write to it can be found in the ELF dumps:

```
80000224 <pass>:
80000224:	0ff0000f          	fence
80000228:	00100193          	li	gp,1
8000022c:	00000073          	ecall

...

80000004 <trap_vector>:
80000004:	34202f73          	csrr	t5,mcause
80000008:	00800f93          	li	t6,8 // User interrupt
8000000c:	03ff0a63          	beq	t5,t6,80000040 <write_tohost>
80000010:	00900f93          	li	t6,9 // Supervisor Interrupt
80000014:	03ff0663          	beq	t5,t6,80000040 <write_tohost>
80000018:	00b00f93          	li	t6,11 // Machine interrup
8000001c:	03ff0263          	beq	t5,t6,80000040 <write_tohost>

...

80000040 <write_tohost>:
80000040:	00001f17          	auipc	t5,0x1
80000044:	fc3f2023          	sw	gp,-64(t5) # 80001000 <tohost>
80000048:	ff9ff06f          	j	80000040 <write_tohost>
```

However I can't find the source code for these functions anywhere. Is there a script somewhere that defines these and builds all the tests?

allenjbaum, 2023-01-19:
Note that this is not in the riscv/riscv_arch_test directory, i.e.
riscv-tests are not the "official" architectural compatibility tests.

Timmmm, 2023-01-19:
Ah you mean [this](https://github.com/riscv-non-isa/riscv-arch-test)? Interesting! Can you run those tests on this Sail model? I presume it must be possible because [this page](https://github.com/riscv-non-isa/riscv-arch-test/tree/main/riscv-test-suite) says the test signatures are generated using Sail (though they seem to have removed all the instructions for how to do it; not sure what is going on there).

pawks, 2023-01-19:
You can find the instructions [here](https://riscof.readthedocs.io/en/stable/arch-tests.html). You will find detailed instructions to setup the environment [here](https://riscof.readthedocs.io/en/stable/installation.html). Note that it is expected that the tests are run on two different models and the signatures are compared by default. So you can either run the tests by using SAIL as both DUT and reference model or use the `--no-dut-run` flag to not run the tests on the DUT(this skips the signature checking too).

billmcspadden-riscv, 2023-01-19:
The existing .elf tests were compiled sometime in the distant past
and the binaries were committed to the repo.  I was unable to track
down the source files and the compilation flow that gave us the binaries.
This is,  of course,  a problem.   And we'll fix this with the following
method:

I have the following goals for 2023:

   - Run the ACTs on the Sail model and extract the signature for each test.
   - Run the ACTs on Spike and extract the signature for each test.
   - Compare signatures
   - Integrate this methodology into the Sail model CI.
   - This effort will also include the configuration of the models using the
   RISCV-Config YAML structures (so that WARL fields, etc are configured
   correctly.

Comments, as always, are appreciated.

Bill Mc.

jrtc27, 2023-01-19:
The tests were built from https://github.com/riscv-software-src/riscv-tests/tree/master/isa which uses (via a git submodule) https://github.com/riscv/riscv-test-env, which is where you’ll find the HTIF code. There is no customisation needed, the default config works (or at least used to and I assume still does).

jrtc27, 2023-01-19:
Question answered and no undocumented customisation

closed 2023-01-19 by jrtc27

Timmmm, 2023-01-19:
Amazing, thanks for your help!