riscv/sail-riscv #209: Needed:  command line switch that allows for experimental functionality of model

billmcspadden-riscv, 2023-01-31:
For RISC-V extensions that have Sail implementations,  but whose specs have not been ratified,  we need a command line switch that enables "experimental" execution of these extensions.  

Background:
In the discussion within the tech-golden-model group,  two schemes were proposed to support these types of extensions:  1) to use a git "feature branch" upon which the extension would be included,  and 2) follow the pattern of the LLVM compiler which uses a command line switch to enable the experimental functionality.  The consensus of the group was to use #2.

billmcspadden-riscv, 2023-02-06:
The LLVM experimental switches that are used for this sort of thing are (per email with Philipp Tomsich):

See [LINK](https://llvm.org/docs/RISCVUsage.html#experimental-extensions) for details.

Short summary:
  -menable-experimental-extensions + "experimental-<extension>" (in the march= string)

jordancarlin, 2025-04-11:
Closed with #800

closed 2025-04-11 by jordancarlin