riscv/sail-riscv #212: Upgrade planning for Sail 0.15

marnovandermaas, 2023-02-09:
@bacam @rmn30 @jrtc27 @billmcspadden-riscv
The question of how to guide the coevolution of Sail and the RISCV model is interesting and important.

I think we should strive to keep both in sync, if this is possible at all. This probably needs a bit of structure.

- Sail team should ideally inform RISCV of anticipated breaking changes.
- It would be ideal to think about tool support.
- How far behind the bleeding edge should the RISCV golden model be?
- Who is responsible for adapting older	Sail?
- How to update the CI
- How do we validate updated Sail?

I think	Robert's suggestion to have a bleeding edge branch of the model	that is	used to	update the model is a good starting point. I think this needs a bit of though and planning, otherwise we might end up in a mess!

_Originally posted by @martinberger in https://github.com/riscv/sail-riscv/issues/157#issuecomment-1354655890_

martinberger, 2023-02-09:
I think we should have a brief note / link somewhere prominent that informs RISCV model users of our policy.

allenjbaum, 2023-02-10:
> #157 (comment)

In my opinion, the top two priorities of any new Sail model are:
- run time configurability (which is being worked on)
 - easy model build without dependencies that require sudo access (or much
of anything not included in standard distros,
because there are far too many users that aren;t allowed to install random
packages that aren't pre approved by IT, and certainly don't have sudo
privs.

martinberger, 2023-02-10:
> easy model build without dependencies that require ...

Hello @allenjbaum, could you be more specific about your second point? 

It seems to be trivial to spin up a cloud instance to run the model outside of the confines of a restricted network. A company doing processor design should also be able to pay for a physical machine running in a corner somewhere, potentially not connected to the internet (after downloading the relevant software) and running sudo. So presumably there is more to this requirement, than meets my eye? Could you bullet-point the core requirements that make this non-trivial?

Alasdair, 2023-02-10:
Docker containers seem to be the de-facto standard solution for packaging something together with all its dependencies these days, although my understanding is rootless docker has some limitations still

abukharmeh, 2023-02-10:
I think we are mixing between an upgraded RISC-V Sail model and updating the model to match any changes in latest compiler revision (for e.g #207 )? I think these points are better addressed in a different issue ?

martinberger, 2023-02-10:
@abukharmeh I agree, they are different issues (but not totally unrelated). We should keep them separate. I think we should have @allenjbaum 's second point as a separate feature request with clear-cut requirements. That makes it more likely that we can address them.

allenjbaum, 2023-02-10:
Again: Docker is not supported in some of the most popular distros that are
used by large developer tool distros (I'm pointing at your, RHEL).
And: installing docker requires Sudo access, and we back to square one.
Requiring developers to find alternatives (Podman has been mentioned, and
there is at least one other) could also solve the problem,
but users shouldn't be required to search them out. They need to be able to
just point and shoot, so to speak.
When we finally get a Reference-Signature-as-a-Service going, that will
solve some of the issue, but that does require users to send their core
configuration YAML to the cloud, and that might not be allowed, so back to
square one.

(Having said that: Marc Karasek does have a repo with scripts that enable
Sail to be builtin /usr.home, I think, but I haven't heard a lot of
feedback for people that have tried to use it.
IF those scripts we hosted inside sail-riscv, that would avid some issues
also.

Alasdair, 2023-02-10:
I don't think there's anything that would stop somebody from building everything in their home directory without root starting with GNU Make and a C/C++ compiler. It's just much easier, and I suspect the vastly more common case that somebody can grab z3 and opam from their package manager and go from there. Without root you could also use a non-distro package manager like Nix that installs locally also.

The Z3 releases have binaries that work across most Linux distributions, provided you don't have an ancient glibc (and I've tested Sail back to Z3 4.8 which is itself quite old now). Even then if you have a working C++ compiler it's not hard to build Z3 from source.

Opam has a way to build itself and bootstrap OCaml locally https://github.com/ocaml/opam#compiling-without-ocaml, although I've never needed to use it myself so I don't know if there are any pitfalls there.

bacam, 2023-02-10:
Opam also has a Z3 package that doesn't need root, and Z3 itself doesn't appear to have any unusual dependencies.

allenjbaum, 2023-02-11:
We have users that have reported otherwise. Either because the can't use
the package managers you use, or have different distros and don't know they
have to use some specific package manager or have IT policies that
prevent....something.
You can't conclude that because you can do it, everyone can, or everyone
knows how to navigate and pull in all the dependencies, has the required
versions, are allowed to even download all the required versions. We have
had multiple issues reported - they just aren't able to build the Sail
model. E.g. arch-tests issue 290:

Allen, thanks for explanation, our main obstacle is reluctance of IT to
install all these service packages.
I couldn't make them to install opam, for example .
Another roadblock is Linux versions.
We have 7.x something and there is no way I can make them to switch to
something more modern.

He is not the only one that has had issues. This is not an imaginary
problem.
 I can't find the email where z3 was download, but one of the packages it
pulled in needed sudo privileges., but htere wa one.

Alasdair, 2023-02-11:
> You can't conclude that because you can do it, everyone can, or everyone knows how to navigate and pull in all the dependencies, has the required versions, are allowed to even download all the required versions. 

Sorry, I don't see what I can really do other than suggest potential technical solutions. If the problem is social I don't think I can help. Maybe others will have ideas.

PeterSewell, 2023-02-11:
Hi Allen,

> We have users that have reported otherwise. Either because the can't use
> the package managers you use, or have different distros and don't know they
> have to use some specific package manager or have [...]

I think we need more clarity on the constraints that you think potential
users are
working under.

If some are not allowed to download and locally build _any_ software, then
it's
hard to see how we can help them - but how do such users (e.g.) run Spike?

For potential users who want to run the Sail model but are not writing any
Sail
(i.e., not extension authors), one could have automation that automatically
builds the C emulator for every commit to the model, and provides that
somehow.

Peter

martinberger, 2023-02-11:
@allenjbaum Is it the best use of time of the volunteers here, some of whom are world leading verification experts, to debug corporate sys-admin problems?

allenjbaum, 2023-02-14:
Well it depends. If we want the Risc-V Sail CSIM to be widely adopted, it
has to be accessible.
People seem to have no problem building Spike; the same people seem to have
a lot of problems building Sail and the Csim model.
It is one thing to make it work for ARM; it's another to make it work at
scale, and that's where it seems to be failing.
(riscof seems to similar problems, by the way)

That is not merely a social problem that can be hand-waved away; IT
organizations don't want to allow uploading and running random tools;
They don't like to install new versions even (e.g. look at distros that a
stuck on specific, old OS and Python versions)
That's not merely a social problem. It's a corporate governance  and
liability problem and you don't get to wave your hands and make it
magically go away.
The downside risk of installing unvetted tools can be considerable.
Like, millions and billions of dollars/pounds/Euros considerable for big
companies, and bankruptcy for smaller ones.

The easy fix is what I term "Reference Signature as a Service" where we let
people log into a site, upload their YAML, and download their signatures.
Bill is working on that (and a bunch of other stuff). There may be
companies that don't want to ship their YAML, though, and that is something
we will have to work through.

PeterSewell, 2023-02-14:
Hi Allen,

> Well it depends. If we want the Risc-V Sail CSIM to be widely adopted, it
> has to be accessible.
> People seem to have no problem building Spike; the same people seem to have
> a lot of problems building [...]

> That's not merely a social problem. It's a corporate governance and
> liability problem and you don't get to wave your hands and make it
> magically go away.

I'm not sure that's very helpful phrasing.

We are obviously not in a position to do anything about corporate IT
policies.
We would obviously like to make the Sail infrastructure as easy as we can
(given our very limited available resources) for the target audience to use,
but any technical change (if any such is possible) needs actionable
facts about why it is hard.  Using Spike also needs installation of new
software, so why _specifically_ is that feasible for people while you think
installing Sail is not?

Peter

martinberger, 2023-02-14:
> IT organizations don't want to allow uploading and running [...] The easy fix is what I term "Reference Signature as a Service" where we let people log into a site, upload 

Isn't there an immediate contradiction between the "easy fix" and what the organisations don't want?

bacam, 2023-02-14:
One aspect that isn't helping is that the requirements of these particular users is unclear.  My impression is that they don't need to interact with the Sail tools at all - instead they just need the generated C simulator.  If that's the case you could package up the intermediate C sources together with the library dependencies (the repo already includes the only non-standard library, but that doesn't mean the others are installed) and a simple shell script to build it.

martinberger, 2023-02-14:
> 

Whoops, sorry @PeterSewell It seems I accidentally rewrote your comment, rather than replying. I'm not sure how to undo this.

pawks, 2023-02-14:
In my experience, only the dependencies required to install OPAM and SAIL required sudo access. So if there is a way to package the [self-sufficient] binary and list it as releases in the repo, it would definitely make all the issues go away. The releases can just include the binaries for the most common distros out there such as 
- Debian (Ubuntu)
- Redhat
- Centos

allenjbaum, 2023-02-14:
Peter asked a question I've also asked one specific user: why can't you
install and run Sail, but you can install and run Spike?
I don't think I got a clear answer, but part was the sudo access, and part
was that Spike seems to be blessed by RVI, but not Sail.
The Sail model is  - but not the Sail language and compiler is how I
interpreted, while the compiler for Spike is C++ which is in the distro
he's using.
That doesn't sound like an easily solvable problem (though it might look
good for Sail if it were a standard supported compiler),
and it's not something I think we should spend effort solving, as opposed
to effort getting around it.

It's unclear to me if providing a binary for each distro would make the
problem go away or not, but likely would since the only data it needs to
access are the riscv test repos
(arch-test, riscof, riscv-config, and maybe isac?t), and their YAML file.

martinberger, 2023-02-14:
> I don't think I got a clear answer, but part was the sudo access, and part was that ...

- Putting the Sail-derived Csim in the cloud solves the sudo problem, but not the *"confidental information should not leave the company's intranet"* problem. 
- Peter's suggestion to provide the C sources for the  Sail-derived Csim solves  the sudo and the *"confidental information should not leave the company's intranet"* problems.

Maybe you can ask the companies in question to post a precise description of their problems here.
Peter's suggestion to supply the source C code for local compilation gets around the

jrtc27, 2023-02-14:
> > Peter asked a question I've also asked one specific user: why can't you install and run Sail, but you can install and run Spike?
> 
> An analysis of the two in terms of dependencies.
> 
> ## SPIKE
> * Has no/minimal dependencies
>   
>   * Device tree compiler: I have no idea whether this infringes the corporate guidelines. But I believe any system which compiles and runs custom binaries would already have it.
>   * RISCV Toolchain: Binaries are available.
> * Dependency tools needed to compile: gcc(already present)
> 
> ## SAIL Model
> * Has multiple package dependencies: `build-essential libgmp-dev z3 pkg-config zlib1g-dev`

build-essential is a Debian meta package for GCC, glibc headers, make and dpkg-dev. The first three are needed for pretty much any open-source software, and the final one isn't needed for anything here. Not sure where you got this list from either then.

As for z3, opam can install it.

>   * Z3 has binary available
>   * Others are the same dependencies one would need to compile the RISCV toolchain. So if the toolchain is locally built, they may already be present.
> * Requires multiple dependent tools to be installed: Opam(ocaml), SAIL(the language).
>   
>   * While both opam and SAIL may be compiled from source if the mentioned package dependencies are already installed, it involves additional steps which might be a factor.

opam can do the compiling from source for you. Installing opam as an unprivileged user is a case of running the one-line
```
bash -c "sh <(curl -fsSL https://raw.githubusercontent.com/ocaml/opam/master/shell/install.sh)"
```
from https://opam.ocaml.org/doc/Install.html and pointing it at a directory you can write to (if the directory doesn't exist, create it first, it seems the script assumes it needs sudo in that case). If you can't install bubblewrap then you'll need to pass --disable-sandboxing to opam init, too, but that's it.

So the only reason people wouldn't be able to build the model themselves is if they're severely restricted by policy in terms of what software they're allowed to download and run as an unprivileged user. But that wouldn't be because of any technical block, that'd be purely a social rule that'd been decreed that they are expected to follow like any other "don't do X". And there's not a lot you can do if companies want to shoot themselves in the foot like that.

martinberger, 2023-02-14:
@pawks 

It sounds like a script that compiles everything in the Sail toolchain from source would be helpful.

Clearly we cannot provide an integration with the full internal toolflow.

jrtc27, 2023-02-14:
> @pawks
> 
> It sounds like a script that compiles everything in the Sail toolchain from source would be helpful.
> 
> Clearly we cannot provide an integration with the full internal toolflow.

You don't need such a thing, and shouldn't use it, either, you should be using opam, whether system or personal, to install Sail...

Alasdair, 2023-02-14:
> build-essential is a Debian meta package for GCC, glibc headers, make and dpkg-dev. The first three are needed for pretty much any open-source software, and the final one isn't needed for anything here. Not sure where you got this list from either then.

We support compressed ELF files in the generated C emulator, that's where the zlib dependency comes from. Could in theory be optional, but zlib is so ubiquitous it's never been an issue.

martinberger, 2023-02-14:
> You don't need such a thing, and shouldn't use it, either, you should be using opam, whether system or personal, to install Sail...

Yes and no.

The root  cause of the problem is that we are talking about people who don't use opam and ocaml, in fact have likely never heard about it. They cannot simply tweak opam as you suggest without spending a significant amount of time on the problem. The opam manual is quite intimidating to a newcomer. A script is a temporary measure to get people started. (Compare e.g. `make`, nobody learns using `make` by reading the manual first, instead you start by tweaking existing makefiels. Whether this is how it should be, we can have a long discussion ...)

jrtc27, 2023-02-14:
> Yes and no.
>
> The root cause of the problem is that we are talking about people who don't use opam and ocaml, in fact have likely never heard about it. They cannot simply tweak opam as you suggest [...]

Why? You run the magic one-liner, answer the question about where to put it, and off you go. You do not need to know anything about OCaml.

martinberger, 2023-02-14:
> Why? You run the magic one-liner, answer the question about where to put it, and off you go. 

If the magic one-liner works, put it in a script, so the user doesn't even need to know about `opam` existence at the start.

jrtc27, 2023-02-14:
```
jrtc27@1309053465be:~$ mkdir -p ~/.local/bin
jrtc27@1309053465be:~$ bash -c "sh <(curl -fsSL https://raw.githubusercontent.com/ocaml/opam/master/shell/install.sh)"
## Downloading opam 2.1.4 for linux on x86_64...
## Downloaded.
## Where should it be installed ? [/usr/local/bin] ~/.local/bin
## '~/.local/bin' resolves to '/home/jrtc27/.local/bin', do you confirm [Y/n] 
## opam 2.1.4 installed to /home/jrtc27/.local/bin
## Remember to add /home/jrtc27/.local/bin to your PATH
## Run this script again with '--restore ' to revert.
jrtc27@1309053465be:~$ export PATH="$HOME/.local/bin:$PATH"
jrtc27@1309053465be:~$ opam init --disable-sandboxing
...
jrtc27@1309053465be:~$ opam install sail z3
...
```

I don't think it's unreasonable to expect people to be able to do that. It's not exactly rocket science, and opam tries to be user-friendly.

pawks, 2023-02-14:
>  Not sure where you got this list from either then.

I got the list from [here](https://github.com/rems-project/sail/blob/sail2/INSTALL.md).

jrtc27, 2023-02-14:
> > I don't think it's unreasonable to expect people to be able to do that.
> 
> I agree they should.
> 
> But clearly they do not, otherwise we would not be having this discussion, and the Sail model would not have a reputation of being hard to use. Let me give you an example of the difficulty. Above you use `opam init --disable-sandboxing`, but the `--disable-sandboxing` option is not listed on https://opam.ocaml.org/doc/Manual.html which is where Google leads you with the search term `opam manual`.

But how much of that is just a documentation gap? Have a quick start guide that's basically what I wrote and leave it at that. The trouble with a lot of the other documentation is it's catering to a different audience.

Alasdair, 2023-02-14:
If you use `opam init` by itself on a system that doesn't support sandboxing then it'll link this page https://opam.ocaml.org/doc/FAQ.html#Why-does-opam-require-bwrap in your terminal, which explains how to disable it.

Alasdair, 2023-02-14:
By the way, could any people actually using RHEL 7 tell me what version of GCC they have?

It seems CentOS 7 comes with GCC 4.8... which is not ideal because it doesn't fully support C11

That said I was just able to build and install Sail on CentOS 7 following the instructions on the opam site, followed by `opam install sail` using OCaml 4.08, which is the oldest OCaml we still support. Other than the basic dev tools package, I only needed to yum install gmp-devel and zlib-devel.

martinberger, 2023-02-14:
> *I was just able to build and install Sail on CentOS 7 following* 

Great. Can we make a script out of that, like `install-on-clean-centos7.sh`? Then we can just point CentOS users to that script and say, if it doesn't work, you are not using a clean  CentOS 7?

Alasdair, 2023-02-14:
```
## Requires yum to install:

# Minimal CentOS 7 install does not have any compilers or development tools
yum group install "Development Tools"

# We need zlib and gmp, which are widely used C libraries
yum install gmp-devel zlib-devel

## Now as a regular user

# Make a directory to install in, for example:
mkdir ~/.local/bin

# Select the directory chosen, when asked.
bash -c "sh <(curl -fsSL https://raw.githubusercontent.com/ocaml/opam/master/shell/install.sh)"

# Can choose whether to modify bashrc. Sandboxing would require bwrap.
opam init --disable-sandboxing --compiler=4.08.0

# Update shell environment
eval $(opam env)

# Now install Sail
opam install sail

# Now we need a version of Z3 that works with a prehistoric glibc and C++ compiler, this ubuntu 14.04 version seems to work:
curl -L https://github.com/Z3Prover/z3/releases/download/z3-4.8.1/z3-4.8.1.016872a5e0f6-x64-ubuntu-14.04.zip > z3-old.zip
unzip z3-old.zip
cp z3-4.8.1.016872a5e0f6-x64-ubuntu-14.04/bin/z3 ~/.local/bin
```

Not sure how well it would work as a script, as a few of the commands prompt you for information. It might be best to run them one by one.

I was able to install without even `gmp-devel` or `zlib-devel` with some CFLAGS/LIBRARY_PATH and opam black magic, but I would *highly* recommend avoiding this.

For Z3 I just had to go back in time until I found an executable that was linked against a suitably old glibc.

abukharmeh, 2023-02-15:
@allenjbaum I suspect the reason you mentioned this issue is https://github.com/riscv-non-isa/riscv-arch-test/issues/290#issuecomment-1293942192  Which If we had precompiled models, then people should have no issues generating sigs for RISCOF ?

allenjbaum, 2023-02-15:
> mentioned this issue is riscv-non-isa/riscv-arch-test#290 (comment)

Yes - but he was not the first, shall we say.

jordancarlin, 2025-03-07:
Seems like this has been resolved with newer versions of Sail and binary releases.

closed 2025-03-07 by Timmmm