riscv/sail-riscv #213: build fail on RHEL

aprnath, 2023-02-10:
Build is failing on RHEL 7 system with the error:

```
Error:
SMT solver killed by signal -10
make: *** [generated_definitions/c/riscv_model_RV64.c] Error 1
make: *** Deleting file `generated_definitions/c/riscv_model_RV64.c'
```

Details are:

```
% opam --version
2.1.4
```

```
% sail -v
Sail 0.14 (sail2 @ opam)
```

```
% ocaml -version
The OCaml toplevel, version 4.07.1
```

Any clues on how to proceed?

Note that running with `SAIL_FLAGS=-dsmt_verbose` shows this before the error:
```
Prove (jib/c_backend.ml/201) (0 <= 'ex212734# & 'ex212734# <= (2 ^ 'n - 1)) |- 'ex212734# <= 9223372036854775807
SMTLIB2 constraints are: 
(push)
(declare-const v1 Int)
(declare-const v0 Int)
(define-fun constraint () Bool (and (not (<= v1 9223372036854775807)) (and (<= 0 v1) (<= v1 (- (^ 2 v0) 1)))))
(assert constraint)
(check-sat)
(pop)

Error:
SMT solver killed by signal -10
make: *** [generated_definitions/c/riscv_model_RV32.c] Error 1
make: *** Deleting file `generated_definitions/c/riscv_model_RV32.c'
```

Alasdair, 2023-02-10:
What version of z3 do you have (`z3 --version`)?

The packages on RHEL can be be very old, so I'd guess getting a newer version from https://github.com/Z3Prover/z3 might fix your issue

himanshurawal, 2023-02-10:
This is the version we are using.
% z3 --version
Z3 version 4.12.2 - 64 bit

jrtc27, 2023-02-10:
> This is the version we are using.
> % z3 --version
> Z3 version 4.12.2 - 64 bit

The latest release is 4.12.1, so you must have built a random development snapshot. What exact commit did you build? Perhaps you're hitting the issue fixed by https://github.com/Z3Prover/z3/commit/3712cbdbfd61237779219c49d06fb69e4d77f852?

himanshurawal, 2023-02-10:
It is failing with the current git head.

Commit cloned for build.

commit 02d48adae556ca0e472ec19dfc1dab414e2e55b1 (HEAD -> master, origin/master, origin/HEAD)
Author: Nikolaj Bjorner <nbjorner@microsoft.com>
Date:   Wed Feb 8 08:24:33 2023 -0800
    fix #6573

Alasdair, 2023-02-10:
I see a few commits to the Z3 repository changing how constraints with 2 ^ x are handled over the past two weeks or so. If you pass the -dsmt_verbose constraints directly to the solver what happens?

Alasdair, 2023-02-10:
If you use the latest released version 4.12.1 I think it will work

himanshurawal, 2023-02-10:
I was able to build with version 4.12.1

aprnath, 2023-02-11:
Thanks for the pointers and help.

closed 2023-02-11 by aprnath